keyboardio/Chrysalis #987: Secondary action erases most of layer 0

lassepetersen, 2022-07-07:
**Describe the bug**
I am trying to assign a secondary action (shift) to the J key. When applying the change, most of layer 0 is changed to blanks. Bundle attached in bottom of issue.

**Screenshots**
[image]
[image]

**Desktop (please complete the following information):**
 - OS: macOS Monterey (12.3.1)
 - Chrysalis Version: 0.10.3
 
**Chrysalis Bundle**
[chrysalis-debug.bundle.json.zip](https://github.com/keyboardio/Chrysalis/files/9064971/chrysalis-debug.bundle.json.zip)

algernon, 2022-07-07:
There's a couple of interesting things in the debug log, the most curious one of them is that there are a few requests for which the response does **not** include a `response` field (suggesting it is null). One such command is the `keymap.custom` during the save: all the correct data gets sent, but we get a null response. The next `keymap.custom` query returns a half-cleared keymap.

The first two commands to fail this way are ids `21` and `22` (`keymap.onlyCustom 1` and `keymap.custom <...>`), both of these return a null response. After `keymap.custom`, the next command we send is a `macros.map`, which gets a correct non-null response. Then, our next query, `keymap.default` returns correctly, and the `keymap.custom` query returns data too - but not the data we were supposed to save. It has 32 transparent values, a curiously "round" number.

No conclusions yet, apart from very clearly seeing the bug from the logs, I'll keep looking.

algernon, 2022-07-07:
Turns out that the number of keys mangled was a huge clue towards the root of the problem: the Atreus firmware tries to allocate more storage space than there's EEPROM for. By about 69 bytes.

What happens in this case is that we end up overwriting the start of EEPROM, which in this case happens to be `EEPROMSettings` (4 bytes) and `EEPROMKeymap` (880 bytes). Of these, we overwrite all of `EEPROMSettings`' storage, and the first 65 bytes (or 32.5 keys) of `EEPROMKeymap`.

The solution is new firmware that doesn't overflow.

algernon, 2022-07-07:
Updated firmware can be found in the `firmware-files.tar` asset of https://github.com/keyboardio/Chrysalis-Firmware-Bundle/releases/tag/v0.10.4. The updated firmware will be part of the next Chrysalis release (0.10.4), which I will be cutting later today.

closed 2022-07-07 by algernon