keyboardio/Chrysalis #988: "migrate" button does not provide feedback

hakamadare, 2022-07-08:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Attach a Keyboardio Atreus
2. Launch Chrysalis 0.10.4 on macOS 12.4
3. Select Atreus, click Connect
4. Error message "We found legacy keys on the keymap that are no longer valid. To migrate to the new codes, please press the Migrate button." appears
5. Click MIGRATE
6. Error message remains, there's no visual feedback
7. Issue recurs after quitting Chrysalis and relaunching

**Expected behavior**
I expect to see some visual indication that the "MIGRATE" button does anything; I'm not sure that I successfully clicked it, it doesn't invert color, or flash, or do anything upon mouseDown or mouseUp events.

I expect to see some feedback after the key code migration process finishes, that either says "success" or "failure", and in the event of a failure, I expect to see an error message.

**Screenshots**
[image]
[image]

**Desktop (please complete the following information):**
 - OS: macOS 12.4
 - Chrysalis Version: 0.10.4 (also occurred on 0.10.3 and 0.10.2)

**Additional context**
My firmware is out of date, I haven't been successful in updating it; I'm opening a separate ticket to track that issue.
[chrysalis-debug.bundle.json.zip](https://github.com/keyboardio/Chrysalis/files/9074237/chrysalis-debug.bundle.json.zip)

algernon, 2022-07-23:
Turns out this happens because the function that is supposed to run when you click the "Migrate" button never runs, because:

```
react-dom.development.js?61bb:3832 Uncaught Error: Expected `onClick` listener to be a function, instead got a value of `object` type.
    at getListener (react-dom.development.js?61bb:3832:1)
    at accumulateSinglePhaseListeners (react-dom.development.js?61bb:8537:1)
    at extractEvents$4 (react-dom.development.js?61bb:8178:1)
    at extractEvents$5 (react-dom.development.js?61bb:8206:1)
    at dispatchEventsForPlugins (react-dom.development.js?61bb:8298:1)
    at eval (react-dom.development.js?61bb:8508:1)
    at batchedEventUpdates$1 (react-dom.development.js?61bb:22396:1)
    at batchedEventUpdates (react-dom.development.js?61bb:3745:1)
    at dispatchEventForPluginEventSystem (react-dom.development.js?61bb:8507:1)
    at attemptToDispatchEvent (react-dom.development.js?61bb:6005:1)
```

If it ran, the alert would disappear, and the keymap would be modified, with the save button at the bottom getting enabled too. So it turns out: there *is* feedback. If it runs. I just need to fix the code to run the appropriate function instead of erroring out.

closed 2022-07-25 by obra

hakamadare, 2022-07-29:
Chrysalis 0.11.0 resolved this for me, thanks folks!

PR #998 (2022-07-25): Editor: Fix the onClick handler of LegacyAlert
When the LegacyAlert component was lifted out, it was lifted out as if it was a function within the renderer, rather than a component. As such, it was expecting to receive the onClick handler directly, rather than a prop.

As it is a component which receives props, lets make it aware of that, and use `props.migrateLegacy`, so the handler actually gets called.

Fixes #988.
commit a0b73236 (2022-07-25): Editor: Fix the onClick handler of LegacyAlert

When the LegacyAlert component was lifted out, it was lifted out as if it was a
function within the renderer, rather than a component. As such, it was expecting
to receive the onClick handler directly, rather than a prop.

As it is a component which receives props, lets make it aware of that, and use
`props.migrateLegacy`, so the handler actually gets called.

Fixes #988.