keyboardio/Chrysalis #989: unable to update Atreus firmware

hakamadare, 2022-07-08:
**Describe the bug**
I can't flash my Atreus with the latest stock firmware using Chrysalis.

**To Reproduce**
Steps to reproduce the behavior:
1. Connect a Keyboardio Atreus
2. Launch Chrysalis, connect to keyboard
3. Select Firmware Update
4. Click UPDATE 
5. Hold down ESC, click OK
6. Wait for notification of discovery of bootloader
7. Kicked back to connection screen, error message says "Error flashing the firmware"

**Expected behavior**
I expected the firmware update procedure to complete.  In the event of an error, I expect to see some troubleshooting information that leads me towards a resolution.

**Screenshots**
[image]
[image]
[image]

**Desktop (please complete the following information):**
 - OS: macOS 12.4
 - Chrysalis Version: 0.10.4

**Additional context**
This issue intersects with #988, there's a debug bundle attached to that issue

algernon, 2022-07-08:
Based on the logs in #988, the keyboard failed to enter bootloader mode. It likely happens because our flashing process is very finicky about proper timing. Releasing a key a second too soon, or too late, and it all breaks.

We're working on improving the situation (#941), but it will take a bit of time, there's a lot to untangle.

Meanwhile, I'll try to reproduce this on macOS tomorrow, and see if I can find a workaround or temporary fix.

isipka, 2022-07-09:
same problem here

TrueFalcon, 2022-07-13:
I get the error message every time too, but the keyboard always works and Chrysalis always reconnects and according to the update screen, it did update. Windows 11, updated to yesterday. HP Omen 30.
[image]
.

isipka, 2022-07-21:
From the @TrueFalcon screenshot I  just realised that in addition to "Error flashing the firmware" message both @hakamadare and me lack the lnfo about current firmware version
[image]

algernon, 2022-07-26:
We've just merged a pull request that completely overhauls the flashing process, and we hope that this PR fixes this issue aswell.

If all goes well, the snapshot build ([#371](https://github.com/keyboardio/Chrysalis/actions/runs/2743221900)) will finish in about 20 minutes. Once it is done, can any of you give it a try, and see if it fixes the flashing problem you had?

hakamadare, 2022-07-29:
hot diggity!  Chrysalis 0.11.0 was able to flash the firmware on my Atreus, @algernon == best mouse.

a couple of UX notes:

after i select a firmware version and click Update, if not holding Esc, i see a dialog box telling me to hold Esc before clicking Update, with options of Cancel and OK.  it's not clear to me at this point whether i should
* click CANCEL, and then from the previous screen, hold Esc and click Update
* hold Esc and click OK

i think i would prefer that dialog box to offer Update and Cancel buttons, provided it's possible to implement a flow of hold Esc, click Update in the dialog box, and then proceed with the firmware update process.  if that's not possible, please offer only a Cancel button.

also, once i completed the update process, i got an error dialog telling me an error had occurred while flashing; otoh when i tried to repeat the process, the UI showed that i had the latest firmware version installed.  it's not clear to me whether that error was safe to ignore, or what i would do to address it (factory reset and reflash?), or if i could retrieve the error message by creating a debug bundle.

algernon, 2022-07-29:
> a couple of UX notes:
>
> after i select a firmware version and click Update, if not holding Esc, i see a dialog box telling me to hold Esc before clicking Update, with options of Cancel and OK. it's [...]

Yeah, we'll need to update the instructions, and as you suggested, the button labels there, too. The dialog is *always* displayed, by the way, whether you hold Esc or not. And with Chrysalis 0.11, it doesn't matter when you start holding it! When it needs to be held, if it isn't, Chrysalis will tell you (eventually, it first tries to put the keyboard into programmable mode 4 times, and will notify you on the 5th).

> i think i would prefer that dialog box to offer Update and Cancel buttons, provided it's possible to implement a flow of hold Esc, click Update in the dialog box, and then proceed with the firmware update process. if that's not possible, please offer only a Cancel button.

The flow is already there. The way it works in 0.11 is that Chrysalis will continuously try to put the keyboard into programmable mode (that's the step where Esc needs to be held), and it will keep trying until it succeeds. After about 5 tries, it will even notify you to hold Esc.

So the flow in 0.11 is: hit the update button, and press and hold Esc whenever you feel like, Chrysalis will wait for you. Holding the key is necessary to transition from the Bootloader stage to the Flashing stage, and it can be released once Chrysalis is on the Flashing stage.

> also, once i completed the update process, i got an error dialog telling me an error had occurred while flashing; otoh when i tried to repeat the process, the UI showed that i had the latest [...]

The error might have been something during restoring the stored settings which Chrysalis saved at the beginning of the process. The detailed error message is included in the debug bundle. If you encounter such an error in the future, please create a bundle and open an issue! :)

Thank you for the thorough explanation of your experience, it's very helpful, I'll open a couple of issues to help us improve it, and make it clearer.

algernon, 2022-07-31:
I'll be keeping this open until we can figure out the root cause of the error mentioned by @hakamadare, or until I manage to reproduce it. Flashing the firmware successfully is just part of the firmware update process, restoring the stored settings is also an essential part of it, and the process should succeed all the way through.

I have an idea what may have caused the issue (dropping 1 layer from the default Atreus firmware), but will have to verify the theory.

shoan, 2022-08-06:
I had similar issues when upgrading to 0.11.0. In my case, my current firmware was detected as unknown. I fixed it by first flashing 0.10.4 and then flashing 0.11.0

algernon, 2022-09-12:
It turns out that the root cause *was* related to dropping a layer. I totally forgot to follow up and verify, but @obra independently discovered and verified it for another customer. I still don't exactly know *why* this causes a problem, but it does. The current development snapshot has a workaround.

See #1039 for more gory details.

Since we have a cause, and a workaround, I'm closing this now.

closed 2022-09-12 by algernon