keyboardio/Chrysalis #992: Cannot print layout cards

cobblepot1, 2022-07-15:
**Describe the bug**
I'm trying to print the layout cards but when I type ctrl-P in Chrysalis 10.5.351 it disconnects my Atreus so all I can print is the 'connect' screen. Not sure if it's related but I'm also getting frequent 'communication timeout's when trying to connect. I'm connecting the keyboard directly to my laptop, not to a USB hub.

**To Reproduce**
Steps to reproduce the behavior:
1. Select 'layout cards' to display the layouts
2. Type 'control+P'
3. Chrysalis window refreshes, shows the keyboard 'connect' screen, OS print dialog appears, if I print it prints the connect screen.

**Expected behavior**
Window continues to show layout cards, print dialog appears, I print the layout cards.

**Desktop (please complete the following information):**
 - OS: Win 10
 - Chrysalis Version: 10.5.321
[chrysalis-debug.bundle.zip](https://github.com/keyboardio/Chrysalis/files/9119327/chrysalis-debug.bundle.zip)

algernon, 2022-07-26:
I'm having trouble reproducing this. On my Windows VM, printing to a file or a printer works without an issue. I suspect that the problem here is not with printing itself, but with the disconnect. I'll keep looking, though.

cobblepot1, 2022-07-27:
Yes, I agree that it seems there is some reason that ctrl+P is causing a refresh of the connection. Let me know if I can do anything else to help troubleshoot.

dnagir, 2022-09-30:
I also can't print the layout on OSX and getting the red error message `Error encountered while printing.` without any information why 🤷

edysli, 2022-10-03:
I've hit a similar issue on Linux trying to print the layout cards. Hitting `ctrl-P` crashes Chrysalis with:
```
Crashing due to FD ownership violation:
#0 0x55c46cb0ba93 <unknown>
#1 0x55c46cbb919c <unknown>
#2 0x55c46cbb9161 <unknown>
#3 0x7fb467d51ef3 <unknown>
Trace/breakpoint trap
```
- OS: Ubuntu GNU/Linux 22.04 kernel 5.15
- Chrysalis version 0.11.5

edysli, 2022-10-05:
I tried again with Chrysalis version 0.11.6 using the print button, unfortunately the issue is still present. The OS print dialogue briefly appears, then the application crashes.

MacNetron, 2022-10-26:
> I've hit a similar issue on Linux trying to print the layout cards. Hitting `ctrl-P` crashes Chrysalis with:
>
> ```
> Crashing due to FD ownership violation:
> #0 0x55c46cb0ba93 <unknown>
> #1 [...]

I've got exactly the same issue on Linux Pop!_OS.

- OS: Pop!_OS 22.04 LTS
- Chrysalis version 0.11.9

edysli, 2022-10-26:
I've tried with `Chrysalis-pr-1194.AppImage` as you suggested and the crash no longer happens. :) The print dialogue appeared after hitting `Ctrl-P` and it let me print to a file.

MacNetron, 2022-10-27:
Print dialog is showing but everything is completely frozen and unresponsive after that.

```➜  ~/Applications/beta ./Chrysalis-pr-1194.AppImage 
Logging to /home/marco/.config/chrysalis/logs/1666859674647.json
Checking for update
Error: Error: No published versions on GitHub
    at Object.newError (/tmp/.mount_ChrysaiJjInA/resources/app.asar/node_modules/builder-util-runtime/src/index.ts:55:17)
    at GitHubProvider.getLatestVersion (/tmp/.mount_ChrysaiJjInA/resources/app.asar/node_modules/electron-updater/src/providers/GitHubProvider.ts:104:13)
    at processTicksAndRejections (node:internal/process/task_queues:96:5)
    at AppImageUpdater.getUpdateInfoAndProvider (/tmp/.mount_ChrysaiJjInA/resources/app.asar/node_modules/electron-updater/src/AppUpdater.ts:368:13)
    at AppImageUpdater.doCheckForUpdates (/tmp/.mount_ChrysaiJjInA/resources/app.asar/node_modules/electron-updater/src/AppUpdater.ts:385:20)
    at /tmp/.mount_ChrysaiJjInA/resources/app.asar/webpack:/src/main/ipc_autoupdate.js:42:29
    at node:electron/js2c/browser_init:193:551
Error occurred in handler for 'auto-update.check-for-updates': Error: No published versions on GitHub
    at Object.newError (/tmp/.mount_ChrysaiJjInA/resources/app.asar/node_modules/builder-util-runtime/src/index.ts:55:17)
    at GitHubProvider.getLatestVersion (/tmp/.mount_ChrysaiJjInA/resources/app.asar/node_modules/electron-updater/src/providers/GitHubProvider.ts:104:13)
    at processTicksAndRejections (node:internal/process/task_queues:96:5)
    at AppImageUpdater.getUpdateInfoAndProvider (/tmp/.mount_ChrysaiJjInA/resources/app.asar/node_modules/electron-updater/src/AppUpdater.ts:368:13)
    at AppImageUpdater.doCheckForUpdates (/tmp/.mount_ChrysaiJjInA/resources/app.asar/node_modules/electron-updater/src/AppUpdater.ts:385:20)
    at /tmp/.mount_ChrysaiJjInA/resources/app.asar/webpack:/src/main/ipc_autoupdate.js:42:29
    at node:electron/js2c/browser_init:193:551 {
[1666859674647.json.zip](https://github.com/keyboardio/Chrysalis/files/9877503/1666859674647.json.zip)

  code: 'ERR_UPDATER_NO_PUBLISHED_VERSIONS'
}
Crashing due to FD ownership violation:
#0 0x5580119e1b33 <unknown>
#1 0x558011a59bdc <unknown>
#2 0x558011a59ba1 <unknown>
#3 0x7f7e49ab3ef3 <unknown>
```

dnagir, 2022-10-27:
More recent version had resolved it for me on OSX.

bomm, 2022-12-07:
I get a program crash (SIGTRAP) when I try to print the layout card on Ubuntu 22.04.1 LTS with Chrysalis-0.12.0.AppImage.
Sometimes I can see the Print dialog for fractions of a second, sometimes it crashes before showing the dialog.

[879168:1207/223600.451822:ERROR:cursor_loader.cc(116)] Failed to load a platform cursor of type kNull
[879226:1207/223600.640541:ERROR:sandbox_linux.cc(377)] InitializeSandbox() called with multiple threads in process gpu-process.
Checking for update
Update for version 0.12.0 is not available (latest version: 0.12.0, downgrade is disallowed).

When trying to print layout card:

Crashing due to FD ownership violation:
#0 0x55cce79c7a93 <unknown>
#1 0x55cce7a7519c <unknown>
#2 0x55cce7a75161 <unknown>
#3 0x7f3d3b5c8ef3 <unknown>

Program terminated with signal SIGTRAP, Trace/breakpoint trap.

obra, 2024-02-26:
Chrysalis now runs in-browser using WebSerial and WebUSB, so I'm closing out this issue as obsolete.

closed 2024-02-26 by obra