keyboardio/Chrysalis #1003: Key label problem with custom layer names

cobblepot1, 2022-07-25:
**Describe the bug**
The custom layer names feature is great, thanks for implementing. When layer names are customized, they appear in menu dropdowns, but when there is a secondary action for layer shift on a key, the key label does not reflect the custom layer name, instead still reading "layer #1" or whatever number. This actually makes it even harder to learn layers because the name is not reflected on the printed keymaps.

**To Reproduce**
Steps to reproduce the behavior:
1. Customize a layer name.
2. Assign a key a secondary action for layer shift to the custom name layer.
3. See the key label "layer #1/A" or similar

**Expected behavior**
If layer #1 is named "Number" then the key label should read "Number / A" or similar

**Desktop (please complete the following information):**
 - OS: Win 10
 - Chrysalis Version: 0.10.5.357

algernon, 2022-07-25:
The root cause for this is the same as for #1004: Chrysalis not checking if the firmware-side plugin is available. By not checking, some parts of Chrysalis will use the name set, some other parts will use what the keyboard reports (why that is so is a funny story, but it's long, complicated, so I'll spare you the details).

Since without the plugin, the names aren't saved, the parts of Chrysalis that end up re-querying the keyboard will not have the updated names. If the plugin is present, then all is well, because the names are saved, and Chrysalis reads them back properly:

[image]

#1005 will "fix" this, by making the layer names uneditable when the firmware-side support is not there.

algernon, 2022-07-25:
I was wrong, there *is* an additional issue, where the secondary action is misrendered, even when the plugin is present. It's even on my screenshot.

PR #1008 (2022-07-25): Fix custom layer name displays on the Overview & non-Model01/100 devices
When preparing the custom layer names pull request, I seem to have dropped the patch during a rebase that added support for custom layer names to keyboards other than the Model01 and the Model100. The first patch in this series does just that, modifies the keymap renderers to use custom layer names if available.

The second part is a change to the Overview widget on the sidebar, so it renders the label with custom layer names too.

Both together fix #1003.

[image]
commit fa7236d7 (2022-07-25): Editor: Use custom layer names on the Overview too

When formatting labels during the Overview widget's rendering, use custom layer
names, if available.

Together with the previous commit, this fixes #1003.
closed 2022-07-25 by obra