keyboardio/Chrysalis #1004: Custom layer names deleted

cobblepot1, 2022-07-25:
**Describe the bug**
After I exit Chrysalis, the custom layer names disappear.

**To Reproduce**
Steps to reproduce the behavior:
1. Connect Chrysalis to my Atreus
2. Customize a layer name
3. Press Tab to change fields
4. Press the orange "Save Changes" arrow
5. Exit Chrysalis
6. Disconnect and reconnect keyboard
7. Start Chrysalis and connect Atreus
8. Layers have no custom naming and are back to Layer 1, 2, etc.

**Expected behavior**
Should see custom layer names in Chrysalis.

**Desktop (please complete the following information):**
 - OS: Win 10
 - Chrysalis Version: 0.10.5.357

**Additional context**
Atreus firmware 0.10.4+5

algernon, 2022-07-25:
The problem here is that the firmware is missing support for editable layer names. The firmware that ship with Chrysalis do not have the necessary plugin enabled, and due to eeprom-space constraints, only the Model100 firmware will by default. Other devices will need to use custom firmware to enable the plugin, I'm afraid.

Chrysalis should detect the availability of the plugin, and not allow editing names if it is not present.

cobblepot1, 2022-07-25:
Oh, well, that's very disappointing since I was one of the people championing custom layer names and I have an Atreus! Any chance you can look into how I can export a json from Chrysalis that can be uploaded to http://www.keyboard-layout-editor.com/ so I can manually create keyboard maps to help me learn the layout?

algernon, 2022-07-25:
I understand it is disappointing, but instead of trying to export to KLE, I'd offer another solution: I can compile custom firmware for you, with a few (or even just one) layers less available (to free up eeprom space), but with LayerNames enabled. I can't do that for the default firmware, not without breaking backwards compatibility, but I can easily make a custom one.

(Meanwhile, I opened keyboardio/Chrysalis-Firmware-Bundle/issues/18, to evaluate if we could make a rare exception and break backwards compatibility to enable LayerNames for the Atreus aswell.)

algernon, 2022-07-25:
As per https://github.com/keyboardio/Chrysalis-Firmware-Bundle/issues/18#issuecomment-1193589434, we're going to free up a layer in the default Atreus firmware, and will enable LayerNames.

algernon, 2022-07-25:
For testing purposes, until keyboardio/Chrysalis-Firmware-Bundle#17 is merged, I built firmware of the same branch, the Atreus version of it is available in [Atreus-LayerNames.zip](https://github.com/keyboardio/Chrysalis/files/9178296/Atreus-LayerNames.zip).

commit f5fcfada (2022-07-25): Do not allow editing layer names if the plugin is not available

When the `LayerNames` plugin is unavailable, do not allow editing layer names,
because they'd be lost on disconnect.

Fixes #1004.
closed 2022-07-25 by obra