keyboardio/Chrysalis #1027: not able to use layout and colormap editor: communication timeout

qliu-Inscopix, 2022-07-30:
**Describe the bug**

After upgrading the Chrysalis from 0.10 to 0.11, not able to save/load keyboard layout: communication timeout error.

**Desktop (please complete the following information):**
 - OS: windows 10
 - Chrysalis Version: 0.11

**Additional context**
[chrysalis-debug.bundle.zip](https://github.com/keyboardio/Chrysalis/files/9226708/chrysalis-debug.bundle.zip)

Add any other context about the problem here.

algernon, 2022-07-30:
Interesting! There's nothing in 0.11 that should introduce a regression like this. Does saving work with Chrysalis 0.10, with the 0.11 firmware?

(I'm looking at the debug logs meanwhile, to see if I can spot some clues)

qliu-Inscopix, 2022-07-30:
Not with 0.11 firmware. Reverting to 0.10 firmware worked fine with Chrysalis 0.10.

algernon, 2022-07-30:
Mhm, that narrows things down. Does the Chrysalis 0.10 firmware work with Chrysalis 0.11?

qliu-Inscopix, 2022-07-30:
Yes. Also I was using 0.10 experimental firmware and noticed the layer number increased from 0.10 to 0.11.

algernon, 2022-09-12:
Repeating it here too, so it's easy to find: there is currently an issue with the Model 100 firmware which can easily lead to the keyboard locking up when larger amounts of data is sent to it. We're working on a fix, but meanwhile, Chrysalis 0.11.1, and the current development snapshot both have a workaround that should help with saving changes to the keyboard.

tepmnthar, 2022-09-13:
I'm not getting a timeout error every time. But after the timeout error is thrown out, only a restart can unlock the keyboard from freezing(hot plug doesn't work and I'm quite sure there's no unkilled Chrysalis process left). LED control keys work fine on the keyboard when the issue is triggered, just the computer can not receive key strokes. Maybe there's something wrong in the firmware, I'm not sure.

stoburn, 2022-09-22:
> Repeating it here too, so it's easy to find: there is currently an issue with the Model 100 firmware which can easily lead to the keyboard locking up when larger amounts of data is sent to it. We're [...]

What's the workaround for this happening with 0.11.2?

The-Compiler, 2022-09-26:
While some have already mentioned it, for people finding this issue: An upgrade to the 0.90 firmware *should* help with this in most cases, it did for me: [Model 100 firmware update to fix corruption issues - Help and Getting Started - Keyboardio community](https://community.keyboard.io/t/model-100-firmware-update-to-fix-corruption-issues/5553)

(Also, this seems like a duplicate of #1049, or the other way around?)