keyboardio/Chrysalis #1032: Chrysalis 0.11.0 incorrectly displays Shift-modified keys as "C+key"

parke, 2022-08-19:
See attached screenshot.  The screenshot is of Chrysalis version 0.11.0.

[image]

The selected key is configured to generate `Shift + 9` (i.e. `(` a left parenthesis).

However, the keyboard layout displays the key as `C+9`.  Whereas I would expect `(` to be displayed.

For reference, here is how things look in Chrysalis 0.8.6.

[image]

It seems that both `Shift +` and `Control +` are displayed as `C+` in 0.11.0.

The key works as expected (i.e. when pressed it generates a left parenthesis).

Note that many other keys are displayed differently in 0.11.0 as compared to 0.8.6.  Both screenshots are of the same keyboard with the same layout.  All of the differences are due to `Shift`-modified keys being displayed incorrectly in 0.11.0.

My Atreus is running an old firmware - maybe version 0.8.6, I'm not sure exactly.  The configuration of the Atreus is also months old (i.e. I have not yet pushed a new configuration to the Atreus from Chrysalis 0.11.0).

(Aside: Fyi, I was unable to upgrade the Atreus firmware, but that could be due to the atypical chroot environment that I run Chrysalis in, so I'm not ready to report it as a bug yet.)

obra, 2022-08-19:
Can you 1) tell us what host layout you have configured in Chrysalis and 2)
give the latest snapshot a try? I believe this is an issue that @algernon
has put some effort into the C+ issue since then.

parke, 2022-08-19:
> Can you 1) tell us what host layout you have configured in Chrysalis 

In `Preferences` the `Keyboard layout` is set to `English (US)`.  Is that the "host layout"?

> and 2) give the latest snapshot a try? I believe this is an issue that @algernon has put some effort into the C+ issue since then.

The `C+` display issues do appear to be fixed in `Chrysalis-0.11.1-snapshot.96.AppImage`.  So perhaps this issue should be closed?

parke, 2022-08-23:
I'm going to close this issue as I believe it is fixed in `Chrysalis-0.11.1-snapshot.96.AppImage`.

closed 2022-08-23 by parke