keyboardio/Chrysalis #1034: Communication timeout when connecting to Model01

belden, 2022-08-23:
**Describe the bug**
Communication timeout when connecting to Model01

**To Reproduce**
Steps to reproduce the behavior:
1. Start with fresh Ubuntu 22.04
2. Plug in a Model01. (Mine has never had a firmware update and has never been configured using Chrysalis.)
3. `sudo apt install -y policykit-desktop-privileges policykit-1-gnome`
4. Download and run Chrysalis. Fix `udev` issue if needed.
5.  Notice that `Keyboardio Model 01 /dev/ttyACM0` is correctly detected.
6. Attempt to `[Connect]`. The loading view appears.
7. After a few seconds, `Communication timeout` appears. 

**Expected behavior**

Chrysalis should detect the keyboard and allow me to configure it.

**Screenshots**

[chrysalis-debug-bundle-json.txt](https://github.com/keyboardio/Chrysalis/files/9410887/chrysalis-debug-bundle-json.txt)

**Desktop (please complete the following information):**
 - OS: Ubuntu 22.04
 - Chrysalis Version: Chrysalis 0.11.1-snapshot.96

**Additional context**

Possible duplicate of https://github.com/keyboardio/Chrysalis/issues/1027 except:

* I have not used any experimental firmware
* Different keyboard model
* Different host OS
* Issue isn't fixed by rebooting
* I cloned the github repo and built the application against tags `v0.10.3-snapshot` as well as default `origin/master@v0.11.1-snapshot` to check whether this was an issue solely with the release artifact I downloaded; the issue persists on both of those tags.

obra, 2022-08-23:
> I cloned the github repo and built the application against tags v0.10.3-snapshot as well as default ***@***.*** to check whether this was an issue solely with the release artifact I downloaded; the issue persists on both of those tags.

Do you have modem manager installed? it can stomp on any kind of serial communications device

belden, 2022-08-23:
@obra Indeed I do have it installed:

```sh
$ apt search modem | grep installed
modemmanager/jammy,now 1.18.6-1 amd64 [installed,automatic]
$ sudo apt install -y aptitude
$ aptitude why modemmanager
i   ubuntu-desktop-minimal Recommends network-manager
i A network-manager        Recommends modemmanager
```

Removing `modemmanager` doesn't fix it, but `networkmanager` may be doing something weird still. I'll see if a reboot fixes this.

belden, 2022-08-23:
@obra No dice - removing `modemmanager` and rebooting did not allow Chrysalis to issue its initial `help` and `keymap.default` commands.

(Sorry for the close/reopen spam on this issue. Github didn't let me submit the previous comment, but I was able to submit it by closing and reopening.)

algernon, 2022-09-07:
This appears to be a different issue than #1027 and the other communication timeouts that happen on Windows. In the case of #1027 and (as far as I can tell, all other similar windows issues), there *is* communication between the device and Chrysalis, there are multiple commands that go through just fine, even ones with a big payload. It eventually times out, but not at the start.

In this case, from the logs you provided, that doesn't happen. The first `help` times out immediately. Thus, this is likely a different problem.

Can you try if the [bin/focus-send](https://github.com/keyboardio/Kaleidoscope/blob/master/bin/focus-send) tool in the Kaleidoscope repo works? `DEVICE=/dev/ttyACM0 bin/focus-send help` should give you the output of the `help` command. If that works, then we can be reasonably certain that the problem is in Chrysalis. If it doesn't, then it is very likely somewhere else.

belden, 2022-09-08:
@algernon `bin/focus-send` points the finger elsewhere: the command you requested

```sh
$ DEVICE=/dev/ttyACM0 ./focus-send help
```

just hangs. (I cancelled it after 5 minutes of no output.)

algernon, 2022-09-12:
Come to think of it: if the keyboard never had a firmware update, then you're running the original Model01 Firmware, which did *not* have the necessary plugins enabled for Chrysalis to work. So that's the root problem there.

However, Chrysalis should be smarter about this, and rather than erroring out with a timeout, it should land you at a page that explains the situation, and lets you move to the Firmware Update screen. I'll try to figure out why we don't end up doing that in this case... it *used* to work in the past.

algernon, 2022-09-26:
I verified by now that this happens when connecting to a Model01 that lacks the necessary plugins. With the Model100, we arrive to an intermediate page, that will direct us to the firmware update screen. We should do the same for the Model01 too. We were doing that in the past, but it looks like it broke sometime along the way.

obra, 2024-02-26:
Chrysalis now runs in-browser using WebSerial and WebUSB and this functionality has been rewritten, so I'm closing out this issue as obsolete. Please don't hesitate to open a new issue if https://chrysalis.keyboard.io exhibits the same behavior

closed 2024-02-26 by obra