keyboardio/Chrysalis #1035: Wrong modifier displayed on key

fluffy-critter, 2022-08-26:
**Describe the bug**

A key set with the 'shift' modifier is being displayed as 'ctrl' instead

**To Reproduce**

Set a key to press the `\` key with a Shift modifier.

**Expected behavior**

The brief display in the corner should show `Shift+\` and the keyboard overlay should show `S+\`

**Screenshots**

[image]

[image]

**Desktop (please complete the following information):**
 - OS: macOS 12.4
 - Chrysalis Version: 0.11.0

**Additional context**
First time running Chrysalis in a while, and I'd been informed at startup that some legacy keys had to be migrated. I don't think that's relevant to this specific key though.

algernon, 2022-08-26:
This should be fixed on master, can you try the latest snapshot package, please?

fluffy-critter, 2022-08-26:
Ah, so it is, and it actually displays `|` which is even better.

closed 2022-08-26 by fluffy-critter