keyboardio/Chrysalis #1036: Communication Timeout with Model 100

LazerBoard, 2022-09-06:
**Describe the bug**
I'm getting communication timeout errors when using Chrysalis on Windows with a Model 100. Keyboard is now completely unusable as a result

**To Reproduce**
Steps to reproduce the behavior:
1. Open Chrysalis (I'm running portable version on Windows 10)
2. Make a change in any layer and save, or try to flash firmware
3. Result will be a "Communication error" prompt (for firmware, it fails at step 5, which incidentally means I now have only blank layers and I cannot successfully edit them 😅 )

**Expected behavior**
Chrysalis should allow me to save changes normally and update firmware successfully (technically I believe the firware was updated, but EPROM content not restored..)

**Screenshots**
Let me know if needed

**Desktop (please complete the following information):**
 - OS: Windows 10
 - Chrysalis Version: 0.11.0 - portable

**Additional context**
I might be among the first ones to receive a Model 100 since I live in China.. shipping took a couple of days 😄

LazerBoard, 2022-09-06:
One more thing, I'm trying to attach the .json error log generated by Chrysalis but Github is saying the file type is not supported.. I'm a first time GitHub user so might need some guidance there.

algernon, 2022-09-06:
> One more thing, I'm trying to attach the .json error log generated by Chrysalis but Github is saying the file type is not supported.. I'm a first time GitHub user so might need some guidance there.

You can rename it to `.txt`, or zip it up. GitHub should let you upload either way.

LazerBoard, 2022-09-06:
> You can rename it to `.txt`, or zip it up. GitHub should let you upload either way.

Thanks, that worked. log attached to the original post

algernon, 2022-09-07:
This looks very similar to the other reports I've seen, but which I have been unable to replicate (but I do not have a Windows machine, I can only test via a VM, so that might affect things).

I recently pushed an update that upgrades one of the packages Chrysalis uses under the hood, which had fixes that *seemed* relevant to the problem you and others are experiencing.

Can you test if the [current snapshot](https://github.com/keyboardio/Chrysalis/releases/tag/v0.11.1-snapshot) works for you? (it's 0.11.1-snapshot.97 or later)

LazerBoard, 2022-09-08:
Hello, I tried 0.11.1-snapshot.97 but got the same error. It sounds like you're suggesting this is a Windows specific issue, so I will try to boot one of those portable Linux distros on my windows machine and see if it works that way. Been a while since I toyed with Linux, will let you know.

algernon, 2022-09-08:
That would be fantastic, thank you!

retribution2471, 2022-09-08:
I also have a similar issue, Chrysalis would have a "connection time out" error. I tried making it connect again but this time I'm met with just a white screen on Chrysalis.

LazerBoard, 2022-09-09:
Sadly I have to report that I'm getting the same issue in Ubuntu 20.04 (running off of a pen drive).. could it be a hardware issue?? Log attached below:

[chrysalis-debug.bundle-ubuntu.txt](https://github.com/keyboardio/Chrysalis/files/9533148/chrysalis-debug.bundle-ubuntu.txt)

algernon, 2022-09-09:
We've identified a firmware issue just yesterday that results in communication timeouts. We do not have a fix for the firmware yet, but we do have a workaround for Chrysalis. Can you try the latest snapshot (0.11.1-snapshot.98)?

It will make saving things to the keyboard *slow*, but it should no longer time out.

We're working on a firmware fix in the meantime.

algernon, 2022-09-12:
As this seems to be the same issue as #1027, I'm closing this in favour of that. The root cause is a firmware issue (we're working on that!), but the latest Chrysalis release, both stable and snapshot, have a workaround.

closed 2022-09-12 by algernon

LazerBoard, 2022-09-13:
> We've identified a firmware issue just yesterday that results in communication timeouts. We do not have a fix for the firmware yet, but we do have a workaround for Chrysalis. Can you try the latest [...]

Hey, that works! Thanks a lot! Now, is there a way to restore the default layers, since they got completely erased during the firmware upgrade? Or alternatively, can I find somewhere the map of the default layers so I can manually re-create them?

algernon, 2022-09-13:
> Now, is there a way to restore the default layers, since they got completely erased during the firmware upgrade?

Yes! Go into `Preference`, then `My Keyboard`, and there will be a factory reset button at the end. Click that, confirm, and wait 10-15 seconds. Then unplug & replug your keyboard, and the factory settings should be back.