keyboardio/Chrysalis #1037: Shift modifier reports as Control

cceckman, 2022-09-06:
**Describe the bug**

I loaded this layout, where I have keymappings for `{` and `}` in layer 1. Those aren't "real" symbols on the keyboard, so I had them set as `Shift + [` and `Shift + ]`.

I connected to this keyboard with Chrysalis 0.11.0 and went to check the layout. Chrysalis gives a "migrate from legacy prompt" - which I'm ignoring for now.

These keys report as `Ctrl + [` and `Ctrl + ]` in the graphical view and the layer view; but show the "shift" modifier toggled in the modifier panel.

[image]

After hitting the "migrate" prompt, this issue persists.

Toggling the `Shift` modifier off and on again doesn't fix the issue; it still shows up as `Ctrl` in the graphical and layer panels.

Loading the firmware to the device doesn't resolve the issue.=

**To Reproduce**

1. Open a Keyboardio Atreus in Chrysalis 0.11 on Windows 11.
2. Select a blank (transparent) key in the graphical view. Select layer 1 in the layer panel.
3. Assign a Shift-able character to the key, e.g. `P`, using the virtual keyboard
4. In the "modifier" dropdown, toggle `Shift` on.

**Expected behavior**
The graphical view reports `S+P` on the key; the layer view reports layer 1 has `Shift + P` assigned to the key.

**Actual behavior**

[image]

**Desktop:**
I don't have a desktop at the moment:

 - OS: Windows 11
 - Chrysalis Version: 0.11.0
 - Keyboard: Keyboardio Atreus

cceckman, 2022-09-06:
I upgraded the keyboard firmware to 0.11.0; the issue persists.

shirshir, 2022-09-06:
This looks the same as https://github.com/keyboardio/Chrysalis/issues/1035

algernon, 2022-09-07:
It is, indeed, the same as #1035. The current snapshot release should fix the issue. And I should really cut a .1 release soon.

closed 2022-09-09 by algernon