keyboardio/Chrysalis #1049: Model100 connection timeout

LuyeT, 2022-09-17:
**Describe the bug**
Out of about 15 tries, the driver has only managed to connect once, in which case when i clicked on 'layout card'(?) tab, the connection would time out again. Also not sure if this is intended behaviour, but whenever I attempt connection, the keyboard is inoperable until replugging USB cable, this was also the case the one time the connection was succesful. I've confirmed modemmanager is not installed

**To Reproduce**
Steps to reproduce the behavior:
1. Open chrysalis
2. Select keyboard
3. Click on Connect
4. Get connection timeout error
5. Replug USB cable

**Expected behavior**
Connecting to keyboard and proceed to configuration of keyboard

**Desktop (please complete the following information):**
 -  Arch linux-ck-zen-5.12.18-1
 - Sway wm
 -  Chrysalis Version: 0.11.1-1

undu, 2022-09-22:
I'm having the same issue with my model 100, I'm using Chrysalis 0.11.2 appimage in archlinux with default kernel

cdwilson, 2022-09-22:
I'm also seeing the same issue on macOS Monterey with Chrysalis 0.11.2.  Keyboard keeps disconnecting and requires hard power cycle to start working again.

algernon, 2022-09-22:
I'm having trouble reproducing this with 0.11.2, could you perhaps upload a debug bundle, please?

You can create a debug bundle by selecting the "Report a problem" item in the main menu. The bundle includes the logs for the *current* Chrysalis session only, so the best would be to reproduce the problem first, then create & upload a bundle without restarting it.

cdwilson, 2022-09-22:
@algernon let me know if this isn't the thing you're looking for:

```json
{
  "timestamp": "2022-09-22T16:17:09.232Z",
  "uuid": "38bd6535-56f4-4a0f-9f8c-60814faa58e7",
  "chrysalis": {"version": "0.11.2"},
  "os": {"platform": "darwin", "distro": "macOS", "release": "12.6", "codename": "macOS Monterey", "kernel": "21.6.0", "arch": "arm64", "hostname": "MacBook-Pro.wilson.network", "fqdn": "MacBook-Pro.wilson.network", "codepage": "UTF-8", "logofile": "darwin", "serial": "778CC57A-CF0B-3D35-8EE8-5035142D0177", "build": "21G115", "servicepack": "", "uefi": true},
  "usb": [],
  "logs": [
    {"function": "find", "label": "focus", "level": "debug", "message": "serial port list obtained", "portList": [{"path": "/dev/tty.Bluetooth-Incoming-Port"}, {"locationId": "03110000", "manufacturer": "Keyboardio", "path": "/dev/tty.usbmodem65649FD713391", "productId": "0006", "serialNumber": "65649FD71339", "vendorId": "3496"}], "timestamp": "2022-09-22T16:11:04.327Z"},
    {"devices": [{"focusDeviceDescriptor": {"bootloader": false, "info": {"displayName": "Keyboardio Model 100", "firmwareType": "bin", "product": "Model100", "urls": [{"name": "Homepage", "url": "https://shop.keyboard.io/"}, {"name": "Forum", "url": "https://community.keyboard.io/"}, {"name": "Chat", "url": "https://keyboard.io/discord-invite"}], "vendor": "Keyboardio"}, "keyboard": {"columns": 16, "rows": 4}, "usb": {"bootloader": {"productId": 5, "protocol": "dfu", "vendorId": 13462}, "productId": 6, "vendorId": 13462}}, "locationId": "03110000", "manufacturer": "Keyboardio", "path": "/dev/tty.usbmodem65649FD713391", "productId": "0006", "serialNumber": "65649FD71339", "vendorId": "3496"}], "function": "find", "label": "focus", "level": "debug", "message": "supported devices found", "timestamp": "2022-09-22T16:11:04.328Z"},
    {"level": "info", "message": "Connecting to port", "path": "/dev/tty.usbmodem65649FD713391", "timestamp": "2022-09-22T16:11:07.261Z"},
    {"level": "info", "message": "Probing for focus support...", "timestamp": "2022-09-22T16:11:07.282Z"},
    {"label": "focus", "level": "verbose", "message": "request", "request": {"args": [], "command": "help", "id": 0}, "timestamp": "2022-09-22T16:11:07.284Z"},
    {"label": "focus", "level": "verbose", "message": "request", "request": {"args": [], "command": "help", "id": 1}, "timestamp": "2022-09-22T16:11:07.284Z"},
    {"label": "focus", "level": "verbose", "message": "request", "request": {"args": [], "command": "help", "id": 2}, "timestamp": "2022-09-22T16:11:07.284Z"},
    {"label": "focus", "level": "verbose", "message": "request", "request": {"args": [], "command": "help", "id": 3}, "timestamp": "2022-09-22T16:11:07.284Z"},
    {"label": "focus", "level": "verbose", "message": "request", "request": {"args": [], "command": "help", "id": 4}, "timestamp": "2022-09-22T16:11:07.291Z"},
    {"args": ["/editor"], "level": "debug", "message": "navigating to location", "timestamp": "2022-09-22T16:11:07.291Z"},
    {"label": "focus", "level": "verbose", "message": "request", "request": {"args": [], "command": "keymap.default", "id": 5}, "timestamp": "2022-09-22T16:11:07.298Z"},
    {"error": "Communication timeout", "label": "focus", "level": "error", "message": "request timed out", "request": {"args": [], "command": "help", "id": 0}, "timestamp": "2022-09-22T16:11:37.565Z"},
    {"error": "Communication timeout", "label": "focus", "level": "error", "message": "request timed out", "request": {"args": [], "command": "help", "id": 1}, "timestamp": "2022-09-22T16:11:37.566Z"},
    {"error": "Communication timeout", "label": "focus", "level": "error", "message": "request timed out", "request": {"args": [], "command": "help", "id": 2}, "timestamp": "2022-09-22T16:11:37.566Z"},
    {"error": "Communication timeout", "label": "focus", "level": "error", "message": "request timed out", "request": {"args": [], "command": "help", "id": 3}, "timestamp": "2022-09-22T16:11:37.566Z"},
    {"error": "Communication timeout", "label": "focus", "level": "error", "message": "request timed out", "request": {"args": [], "command": "help", "id": 4}, "timestamp": "2022-09-22T16:11:37.566Z"},
    {"error": "Communication timeout", "label": "focus", "level": "error", "message": "request timed out", "request": {"args": [], "command": "keymap.default", "id": 5}, "timestamp": "2022-09-22T16:11:37.566Z"},
    {"level": "info", "message": "Disconnecting from keyboard", "timestamp": "2022-09-22T16:11:37.567Z"},
    {"args": ["./"], "level": "debug", "message": "navigating to location", "timestamp": "2022-09-22T16:11:37.567Z"},
    {"args": ["/keyboard-select"], "level": "debug", "message": "navigating to location", "timestamp": "2022-09-22T16:11:37.583Z"},
    {"function": "find", "label": "focus", "level": "debug", "message": "serial port list obtained", "portList": [{"path": "/dev/tty.Bluetooth-Incoming-Port"}, {"locationId": "03110000", "manufacturer": "Keyboardio", "path": "/dev/tty.usbmodem65649FD713391", "productId": "0006", "serialNumber": "65649FD71339", "vendorId": "3496"}], "timestamp": "2022-09-22T16:11:37.594Z"},
    {"devices": [{"focusDeviceDescriptor": {"bootloader": false, "info": {"displayName": "Keyboardio Model 100", "firmwareType": "bin", "product": "Model100", "urls": [{"name": "Homepage", "url": "https://shop.keyboard.io/"}, {"name": "Forum", "url": "https://community.keyboard.io/"}, {"name": "Chat", "url": "https://keyboard.io/discord-invite"}], "vendor": "Keyboardio"}, "keyboard": {"columns": 16, "rows": 4}, "usb": {"bootloader": {"productId": 5, "protocol": "dfu", "vendorId": 13462}, "productId": 6, "vendorId": 13462}}, "locationId": "03110000", "manufacturer": "Keyboardio", "path": "/dev/tty.usbmodem65649FD713391", "productId": "0006", "serialNumber": "65649FD71339", "vendorId": "3496"}], "function": "find", "label": "focus", "level": "debug", "message": "supported devices found", "timestamp": "2022-09-22T16:11:37.594Z"}
  ]
}
```

cdwilson, 2022-09-22:
Here's what I see in the UI as well:

[image]

undu, 2022-09-22:
When taking this bundle the keyboard hang and was forced to unplug and plug again to make it work:
```json
[
    {"label": "focus", "level": "verbose", "message": "request", "request": {"args": [], "command": "settings.defaultLayer", "id": 11}, "timestamp": "2022-09-22T16:18:07.138Z"},
    {"data": "0 ", "label": "focus", "level": "debug", "message": "incoming data", "timestamp": "2022-09-22T16:18:07.390Z"},
    {"label": "focus", "level": "verbose", "message": "response", "request": {"args": [], "command": "settings.defaultLayer", "id": 11}, "response": "0", "timestamp": "2022-09-22T16:18:07.390Z"},
    {"args": [], "command": "keymap.layerNames", "label": "focus", "level": "verbose", "message": "request (noop)", "timestamp": "2022-09-22T16:18:07.391Z"},
    {"label": "focus", "level": "verbose", "message": "request", "request": {"args": [], "command": "plugins", "id": 12}, "timestamp": "2022-09-22T16:18:07.497Z"},
    {"label": "focus", "level": "verbose", "message": "request", "request": {"args": [], "command": "plugins", "id": 13}, "timestamp": "2022-09-22T16:18:07.497Z"},
    {"label": "focus", "level": "verbose", "message": "request", "request": {"args": [], "command": "plugins", "id": 14}, "timestamp": "2022-09-22T16:18:07.497Z"},
    {"label": "focus", "level": "verbose", "message": "request", "request": {"args": [], "command": "plugins", "id": 15}, "timestamp": "2022-09-22T16:18:07.497Z"},
    {"label": "focus", "level": "verbose", "message": "request", "request": {"args": [], "command": "plugins", "id": 16}, "timestamp": "2022-09-22T16:18:07.497Z"},
    {"label": "focus", "level": "verbose", "message": "request", "request": {"args": [], "command": "plugins", "id": 17}, "timestamp": "2022-09-22T16:18:07.498Z"},
    {"label": "focus", "level": "verbose", "message": "request", "request": {"args": [], "command": "plugins", "id": 18}, "timestamp": "2022-09-22T16:18:07.498Z"},
    {"label": "focus", "level": "verbose", "message": "request", "request": {"args": [], "command": "plugins", "id": 19}, "timestamp": "2022-09-22T16:18:07.498Z"},
    {"label": "focus", "level": "verbose", "message": "request", "request": {"args": [], "command": "plugins", "id": 20}, "timestamp": "2022-09-22T16:18:07.498Z"},
    {"label": "focus", "level": "verbose", "message": "request", "request": {"args": [], "command": "plugins", "id": 21}, "timestamp": "2022-09-22T16:18:07.498Z"},
    {"label": "focus", "level": "verbose", "message": "request", "request": {"args": [], "command": "plugins", "id": 22}, "timestamp": "2022-09-22T16:18:07.498Z"},
    {"label": "focus", "level": "verbose", "message": "request", "request": {"args": [], "command": "plugins", "id": 23}, "timestamp": "2022-09-22T16:18:07.498Z"},
    {"label": "focus", "level": "verbose", "message": "request", "request": {"args": [], "command": "plugins", "id": 24}, "timestamp": "2022-09-22T16:18:07.498Z"},
    {"label": "focus", "level": "verbose", "message": "request", "request": {"args": [], "command": "plugins", "id": 25}, "timestamp": "2022-09-22T16:18:07.498Z"},
    {"label": "focus", "level": "verbose", "message": "request", "request": {"args": [], "command": "plugins", "id": 26}, "timestamp": "2022-09-22T16:18:07.498Z"},
    {"data": "EEPROMKeymap\nColormapEffect\nMacros\nMouseKeys\nMagicCombo\nQukeys\nOneShot\nEscapeOneShot\nPersistentIdleLEDs\nDynamicMacros\n", "label": "focus", "level": "debug", "message": "incoming data", "timestamp": "2022-09-22T16:18:07.751Z"},
    {"label": "focus", "level": "verbose", "message": "response", "request": {"args": [], "command": "plugins", "id": 12}, "response": "EEPROMKeymap\nColormapEffect\nMacros\nMouseKeys\nMagicCombo\nQukeys\nOneShot\nEscapeOneShot\nPersistentIdleLEDs\nDynamicMacros", "timestamp": "2022-09-22T16:18:07.751Z"},
    {"label": "focus", "level": "verbose", "message": "response", "request": {"args": [], "command": "plugins", "id": 13}, "response": "EEPROMKeymap\nColormapEffect\nMacros\nMouseKeys\nMagicCombo\nQukeys\nOneShot\nEscapeOneShot\nPersistentIdleLEDs\nDynamicMacros", "timestamp": "2022-09-22T16:18:07.753Z"},
    {"label": "focus", "level": "verbose", "message": "response", "request": {"args": [], "command": "plugins", "id": 14}, "response": "EEPROMKeymap\nColormapEffect\nMacros\nMouseKeys\nMagicCombo\nQukeys\nOneShot\nEscapeOneShot\nPersistentIdleLEDs\nDynamicMacros", "timestamp": "2022-09-22T16:18:07.756Z"},
    {"label": "focus", "level": "verbose", "message": "response", "request": {"args": [], "command": "plugins", "id": 15}, "response": "EEPROMKeymap\nColormapEffect\nMacros\nMouseKeys\nMagicCombo\nQukeys\nOneShot\nEscapeOneShot\nPersistentIdleLEDs\nDynamicMacros", "timestamp": "2022-09-22T16:18:07.759Z"},
    {"label": "focus", "level": "verbose", "message": "response", "request": {"args": [], "command": "plugins", "id": 16}, "response": "EEPROMKeymap\nColormapEffect\nMacros\nMouseKeys\nMagicCombo\nQukeys\nOneShot\nEscapeOneShot\nPersistentIdleLEDs\nDynamicMacros", "timestamp": "2022-09-22T16:18:07.762Z"},
    {"label": "focus", "level": "verbose", "message": "response", "request": {"args": [], "command": "plugins", "id": 17}, "response": "EEPROMKeymap\nColormapEffect\nMacros\nMouseKeys\nMagicCombo\nQukeys\nOneShot\nEscapeOneShot\nPersistentIdleLEDs\nDynamicMacros", "timestamp": "2022-09-22T16:18:07.775Z"},
[... cut ...] r": "Keyboardio"}, "keyboard": {"columns": 16, "rows": 4}, "usb": {"bootloader": {"productId": 5, "protocol": "dfu", "vendorId": 13462}, "productId": 6, "vendorId": 13462}}, "manufacturer": "Keyboardio", "path": "/dev/ttyACM0", "pnpId": "usb-Keyboardio_Model_100_656451F21439-if00", "productId": "0006", "serialNumber": "656451F21439", "vendorId": "3496"}], "function": "find", "label": "focus", "level": "debug", "message": "supported devices found", "timestamp": "2022-09-22T16:19:22.007Z"}
  ]
```

cdwilson, 2022-09-22:
One thing I just noticed (not sure if this makes any difference), I went to the Firmware Update page and I saw that the current firmware version was listed as "Firmware version unavailable".

[image]

I went through the update process and after the update, the firmware is showing a version now.  

[image]

I'll keep playing around with the app and see if it looses the connection again.

justincjohnson, 2022-09-22:
I'm seeing this issue as well.  Here's a bundle from right after a "Communication timeout" error.
[chrysalis-debug.bundle.json.zip](https://github.com/keyboardio/Chrysalis/files/9628070/chrysalis-debug.bundle.json.zip)

algernon, 2022-09-22:
@cdwilson @justincjohnson Do you happen to have USB Overdrive or something similar running, by any chance?

obra, 2022-09-22:
Or Cura?

cdwilson, 2022-09-22:
@obra no, I don't.

justincjohnson, 2022-09-22:
I'm not sure what that is, but I'm not running anything special afaik.  I do have a Logi Mx Vertical, and therefore running Logi Options.  I saw the same mouse in the demo videos though so I assume that's okay.

obra, 2022-09-22:
Ok!

We're digging into it.

algernon, 2022-09-22:
Still no luck in reproducing this, unfortunately. If any of you could try something for me, that'd help me figure out what's going wrong. Here's what I'd like you to do:

- Install the `focus-send` tool, you can grab a binary for your operating system [from here](https://github.com/keyboardio/focus-send/releases/tag/v0.1.0-snapshot). It's a single binary with no other dependencies.
- Once decompressed, run it with a `help` argument, like this: `focus-send help`
- Let me know here what the output is: a list of commands like `help`, `version`, `keymap.custom`, and a bunch of others, or an error, or a hang followed by an eventual timeout? In either case, I'd love to see the full output of the command, even if it succeeds.

The goal here is to verify whether it is Chrysalis doing something wrong, or if there's something else in play. If two different tools with unrelated code both end up showing the same symptoms, that suggests that the problem is elsewhere. If it works, then it is Chrysalis that's doing something wrong.

undu, 2022-09-22:
Seems broken to me:
```
$ ./focus-send.linux_amd64 help
No device found to connect to
$ ./focus-send.linux_amd64 version
No device found to connect to
$ ./focus-send.linux_amd64 keymap.custom
No device found to connect to

teqdruid, 2022-09-22:
I'm having the same issue, but running Windows 11 22H2. Zipped bundle attached.
[chrysalis-debug.bundle.zip](https://github.com/keyboardio/Chrysalis/files/9629692/chrysalis-debug.bundle.zip)

algernon, 2022-09-22:
The firmware version not showing up before upgrading is as expected, we added the version thing after the factory firmware was finalized.

alexfedoseev, 2022-09-23:
Same here.

macOS 12.6 with Logitech MX Master 3S mouse (Logi Options+)

alexfedoseev, 2022-09-23:
In the Firmware Update, I see this (I'm not sure I should try to update)

[image]

a-gabriella, 2022-09-23:
I am seeing the same issue

pbronez, 2022-09-24:
Similar issue. Tried to update the firmware from factory. Got as far as "hold prog and wait for green lights" and it failed. Now I get timeout errors whenever I try to connect to the keyboard from Chrysalis. Done several physical USB reconnect cycles, no joy.

macOS Monterey 12.6
Chrysalis 0.11.3
Keyboard Model 100

[chrysalis-debug.bundle.json.txt](https://github.com/keyboardio/Chrysalis/files/9637594/chrysalis-debug.bundle.json.txt)

pbronez, 2022-09-24:
> * Install the `focus-send` tool, you can grab a binary for your operating system [from here](https://github.com/keyboardio/focus-send/releases/tag/v0.1.0-snapshot). It's a single binary with no [...]

@algernon  It took some fiddling, but I got this going. Challenge was mostly just convincing MacOS to run the sketchy little unsigned binary. Here's the requested output:

```
sudo ./focus-send help
keymap.custom
keymap.default
keymap.onlyCustom
help
device.reset
plugins
settings.defaultLayer
settings.valid?
settings.vt
plugins
settings.defaultLayer
settings.valid?
settings.version
settings.crc
eeprom.contents
eeprom.free
eeprom.erase
palette
colormap.map
escape_oneshot.cancel_key
idleleds.time_limit
macros.map
macros.trigger
```

tompi, 2022-09-25:
hey, not sure if this is the same problem, but I was struggling for days without managing to flash.

Tried different cables but always timing out. Today I realised I had only tried expensive thunderbolt cables... I tested with the original cable and everything went smoothly... Perhaps the usb controller negotiates a high speed that creates problems on a high end cable?

alexfedoseev, 2022-09-25:
Another thing that might matter is that I connect my Model100 via TP-Link UH700 USB Hub.

pbronez, 2022-09-25:
I'm using the USB-C to USB-A cable that came with the Model 100, connect through the USB-A port on the front of a CalDigit TS3 Plus.

pbronez, 2022-09-25:
I was able to connect the Model 100, update the firmware to 0.90 and flash a few layer updates. Pretty excited that I can get start using the keyboard now!

I'm not sure what changed to make it work this time. I didn't update anything- just tried again randomly after waiting a day or so from the last failure.

alexfedoseev, 2022-09-26:
Same here. Today it works. It was, probably, a weekend already. What are Chrysalis's working days?

The-Compiler, 2022-09-26:
While some have already mentioned it, for people finding this issue: An upgrade to the 0.90 firmware *should* help with this in most cases, it did for me: [Model 100 firmware update to fix corruption issues - Help and Getting Started - Keyboardio community](https://community.keyboard.io/t/model-100-firmware-update-to-fix-corruption-issues/5553)

tlyu, 2022-09-26:
People experiencing cable-related timeouts in Chrysalis: does the keyboard function normally for typing purposes while using the same cable?

LuyeT, 2022-09-27:
So far ive only been trying the cable that came with. Tried a different one
now and connected with driver instantly. The first cable was working fine
for typing.

justincjohnson, 2022-09-27:
> I was able to connect the Model 100, update the firmware to 0.90 and flash a few layer updates. Pretty excited that I can get start using the keyboard now!

I believe I've not had any issues since this upgrade as well.  🥳  Can't remember for sure, but I've definitely not had issues for a while since the upgrade, so things are looking good.  Also I always used the cable that was provided.

Zian, 2022-11-13:
The problem just occurred on my keyboard. Debug bundle is attached. I tried following the link to https://github.com/keyboardio/kaleidoscope-focus.rs/releases/tag/v0.1.0-snapshot but got a 404 page.

I have not changed my cable before the problem started happening. The device is connected as follows:

Starting from the Keyboardio keyboard, one end of the Keyboardio-provided USB cable is plugged in to the keyboard. The other end is plugged into an Anker USB 3 hub. The hub is powered through an external power brick. The hub is connected to a Lenovo Thunderbolt dock using a USB 3.0 cable (not type C).

Switching perspectives slightly, the ThinkPad is connected to the dock by putting the computer down on top of the dock. Then, I slide something from the left side to the right side and hear a nice noise.

[chrysalis-debug.json.txt](https://github.com/keyboardio/Chrysalis/files/9998531/chrysalis-debug.json.txt)

tomako, 2023-07-30:
I have the same issue on Windows 11, however it worked well earlier.
Because I had had another [issue with the automatic upgrade from v0.12.0](https://github.com/keyboardio/Chrysalis/issues/1269) I dropped the old Chrysalis and installed the latest one (v0.13.2). Since that I've been getting the "Communication timeout" messages. I went to another Windows PC - no luck. Another cable was also tested - no luck. v0.12.0 was reinstalled - no luck.
The keyboard itself works.

[chrysalis-debug.json.zip](https://github.com/keyboardio/Chrysalis/files/12209752/chrysalis-debug.json.zip)

obra, 2024-02-26:
Chrysalis now runs in-browser using WebSerial and WebUSB and this functionality has been rewritten, so I'm closing out this issue as obsolete. Please don't hesitate to open a new issue if https://chrysalis.keyboard.io exhibits the same behavior

closed 2024-02-26 by obra