keyboardio/Chrysalis #1054: We need a German QWERTZ layout for the Model 100

algernon, 2022-09-21:
One of the keycap sets available for the Model 100 is German QWERTZ. As such, we should have a layout bundled with Chrysalis that matches those keycaps.

For reference, this is roughly the layout:

https://www.dropbox.com/s/0rne7lic1k6tfza/Model%20100%20QWERTZ%20-%202022-07-07.pdf?dl=0

Some of the symbols were changed / fixed since, but those are just labeling, the positions are the same.

Keep in mind that all layouts bundled with Chrysalis assume an US QWERTY layout on the host side, and German QWERTZ should too. That will, however, show incorrect symbols for some keys.

neuhaus, 2022-09-30:
I received my model 100 and the QWERTZ keycaps and the layout does not match what i see in Chrysalis. In particular, "Ü" was missing on the base layer and some other keys were wrong as well so i corrected them as best as i could.

I am attaching my JSON file with the QWERTZ layout for the model 100.
[Layout Qwertz.json.txt](https://github.com/keyboardio/Chrysalis/files/9682588/Layout.Qwertz.json.txt)

neuhaus, 2022-09-30:
"all layouts bundled with Chrysalis assume an US QWERTY layout on the host side" ... 
I don't like this very much, for example a Macbook will have a QWERTZ builtin keyboard and it makes sense to configure it as QWERTZ on the host so you can use both the external and internal keyboard.

neuhaus, 2022-09-30:
I am attaching a photo of the QWERTZ model 100 as reference
[image]

neuhaus, 2022-09-30:
Are there really that many host side layouts? The QWERTZ layout is fairly standardized.

bomm, 2022-09-30:
It is clear that you cannot manage all possible combinations of keyboard layout and host layout, and I can understand the (current?) limitation to US QWERTY from a programmer's point of view but not from a user's one.

I'm a professional programmer, but unfortunately not used to Javascript. 
I would like to know how exactly the mapping works and how the data is stored. How does the mapping to US QWERTY work for characters that don't exist on a US keyboard, e.g. 'Ä' (&Auml;)?
Can you point me to documentation or specific parts of the source code where I could learn this?

Would it be possible to provide a mechanism that in theory could support all possible combinations, and let the set of supported mappings grow as they are provided by someone? I imagine Chrysalis could show what host layouts are supported by a selected keyboard layout (or vice versa), and maybe encourage users to create missing mappings.

obra, 2022-12-05:
@algernon let's get a QWERTZ layout into Chrysalis ASAP? 

https://community.keyboard.io/t/model-100-german-qwertz-keycap-legends/5381/64?u=rpnfan is one version of the layout. 
@neuhaus' has another version here: https://github.com/keyboardio/Chrysalis/issues/1054#issuecomment-1263313213

I do think we probably need to have two versions of QWERTZ  - one assuming the host OS layout is set to QWERTZ and one assuming it's set to QWERTY

obra, 2022-12-07:
> I agree that the requirement to configure the host layout as US is not a
> good solution. There are good reasons to use a different host layout.
>
> Maybe we should think about a new concept for the definition of keyboard
> layouts.
>
> If I understand correct, the layouts like
> https://github.com/keyboardio/Chrysalis/blob/master/static/Keyboardio/Model100/layouts/qwerty.json
> define the mapping from physical keys to USB HID scan codes. This is what
> the keyboard has to do at the end. The keycodes will be translated by the
> host layout to a symbol, or might get interpreted directly by some software
> in special cases.
>
> Of course it is not feasible to distribute keyboard layout files for many
> Keyboardio/keycaps layouts multiplied by a number of all supported or
> "relevant" host layouts.
>
> *What is the requirement of the user?*
>
> I think in most cases, the user does not care about the scan codes.
> (Probably many don't even know that scan codes exist.) The user wants that
> pressing a key labeled "Y" results in the software displaying a "y" or
> "Y"(*) depending on the modifier keys.
> (*)I call this keysym below.
>
> [... 34 lines cut ...]
> specifically about German layout. It is a general question about using host
> layouts other than US or different host layouts with the same keycaps
> layout.
>
> Bodo

I absolutely did not mean to suggest that we expect German users to
configure their system keyboard layout as US QWERTY. And indeed QWERTZ on a
German QWERTZ host configuration is the priority here.

bomm, 2022-12-07:
> I absolutely did not mean to suggest that we expect German users to configure their system keyboard layout as US QWERTY. And indeed QWERTZ on a German QWERTZ host configuration is the priority here.

My post, although triggered by the latest discussion, was also based on https://github.com/keyboardio/Chrysalis/issues/1054#issue-1381566887: "[...] all layouts bundled with Chrysalis assume an US QWERTY layout on the host side, and German QWERTZ should too." and https://github.com/keyboardio/Chrysalis/issues/1054#issuecomment-1263356884: "There are 203 host-side layouts [...] Even if we'd limit host-side layouts to what is reasonable, we'd still have to translate QWERTZ to about a dozen [...]"

Of course, we can create a German Keyboardio layout file that assumes a German host layout, and more layout files for every combination that anyone is willing to work on. For me it doesn't matter if the files are bundled with Chrysalis or shared somewhere else. This might be the way to get a working configuration soon.

My thought is: Can we invent a better configuration system that would allow defining one Keyboardio layout that would automatically work with more than one host layout? Of course this would be an incompatible change, and modifying Chrysalis or creating an alternative configuration tool will probably need more time. This suggestion for discussion or research is the main point of [my previous comment](https://github.com/keyboardio/Chrysalis/issues/1054#issuecomment-1341001447).

bomm, 2022-12-07:
This is the layout I'm currently using with German keycaps and German host layout.
[Layout-german-bodo-2.json.zip](https://github.com/keyboardio/Chrysalis/files/10180214/Layout-german-bodo-2.json.zip)

rpnfan, 2022-12-18:
Thanks Bodo, I checked your file against the one from Neuhaus and mine. I forgot the ´-key (on the Any-position). In Neuhaus' file there is no way to switch to the Num-layer (#1), which is available in your json on the prog-key. I see no error and nothing missing. So the default German layout file should either be the "plain" file from Bodo (post before this one). 

Optionally one could consider making the two additions I suggested.
1) Layer 0 → add ESC as an additional option on the Prog-key 

As far as I understand there is no price to pay for that, because the Prog-key will work always anyways!? So that would make the transition to the model 100 a bit easier IMO. I think the model 100 default layout should be as close as possible to the original layout. _Then_ everybody can adjust the layout to his own needs, but would not need to do this.

2) allow AltTab by making the  G-key-hold making an additional Alt-key.

This has the disadvantage of going away from a standard keyboard, where holding g will result in gggggggggggggggg… 
But I find it still a good idea to add that. First Alt-Tab is something everybody otherwise would have to find an own solution and therefore I think it is good to take the need to find a solution for that away -- again helping to get started easier. Second this can be a perfect example in the documentation to the German keyboard layout in how the keyboard can be customized.

One could also offer two German layouts. One "plain" and one with additional tricks under the sleeve – of course nicely documented. I am now exploring the options myself. Maybe someone has ideas what could be a great addition besides Alt-Tab functionality?

What do the German speaking people think? Should the documentation for the Germany layout be in German or in English? I am not decided yet what would be better. 

Regards, Peter

nils-a, 2022-12-19:
I'm on my first day of using my Keyboardio, using the json from Peter. (Will check one of the others, later.)

Am I holding it wrong, if I have the feeling the _shift_ and _ctrl_ keys need to be switched?

rpnfan, 2022-12-19:
Hi Nils, maybe better post in the forum instead here? I do not have access to my file. Maybe accidently Shift was changed there accidently compared to the standard position. I suggest you use the file from Bodo (see above) as a starting point. I, like many others, change the Shift either to one position inwards or to the most inner position, because for many the stretch is too large to the outer thumb key for such an often used one.

bomm, 2022-12-19:
I'm unsure about swapping `Ctrl` and `Shift`. _For me_ it is easier to reach the innermost thumb key when pressed alone, but in combination with `fun` it is easier _for me_ to use the outermost one.

I just noticed that ther are no `PgUp` / `PgDn` keys, or did I overlook something?
I'm thinking about moving the `Vol-` / `Vol+` / `Mute` keys to the currently transparent `U` / `I` / `O` in layer 2 and put `PgDn` / `PgUp` to `M` / `,` and maybe `Home` / `End` (also) to `N` and `.`.

bomm, 2022-12-21:
> I'm unsure about swapping `Ctrl` and `Shift`. _For me_ it is easier to reach the innermost thumb key when pressed alone, but in combination with `fun` it is easier _for me_ to use the outermost one.

Now I swapped the `shift` keys with the keys next to them.

> I just noticed that ther are no `PgUp` / `PgDn` keys, or did I overlook something? I'm thinking about moving the `Vol-` / `Vol+` / `Mute` keys to the currently transparent `U` / `I` / `O` in layer 2 and put `PgDn` / `PgUp` to `M` / `,` and maybe `Home` / `End` (also) to `N` and `.`.

I configured this change.

As [proposed](https://github.com/keyboardio/Chrysalis/issues/1054#issuecomment-1356863583) by @rpnfan, I added `Esc` to the `prog` key in layer 0 .

The attached file contains these changes.

[Layout-german-bodo-5.json.txt](https://github.com/keyboardio/Chrysalis/files/10277209/Layout-german-bodo-5.json.txt)

(Attaching screenshots as PNG or JPG doesn't seem to work. I get an error message "Something went really wrong, and we can’t process that file. Try again.")

rpnfan, 2022-12-21:
Hi Bodo, 
thanks for the update. I think that looks fine mostly. What I do not like is the double Home/End-key for a default German layout. I do not comment on what a "best" personal layout could be but I think the default German layout should 
1) include all keys on a standard German keyboard
2a) have these in the similar position if possible
2b) else have them in a logic and/ or good position

For this I would move the CAPS on layer 2 to the normal position of CapsLock (left of A) and remove the Home/End left of A/Y key.

The PrintScreen-function could be place on the P-key, to be easy to remember. It does not belong naturally on the left side (close to the mouse keys), so why not place it so it is easier to remember?

Similiar the INS-key could be at another position. I would choose the I-key, but you have the media keys there already together and I would leave them for that reason. Then the INS-function could be place on the Ö-key, which is the last good position available. Why not use that for the INS-function which in some programs is frequently used? On the X-key (X-tract...) it is even counter intuitive IMO.

I found that Scroll-Lock and Break (Pause)  have been missing. I added those to layer 2 (LED-key and <-key). I do not know if somebody would miss those, but to be complete and close to the standard layout they should be included IMO.

This should make to start using the keyboard as easy and straight forward as possible. What do you and others think?

I attach the new suggestion and just call it Beta6 (after Bodo5 :-)).

[qwertz_Layout_beta6.json.txt](https://github.com/keyboardio/Chrysalis/files/10280850/qwertz_Layout_beta6.json.txt)

I think now everything is fine and nothing missing. If you all agree I wilĺ write the documentation how to use the default Germany layout.

rpnfan, 2022-12-23:
I just noticed that space cadet from Chrysalis does not make sense with the German layout. Left space will give ) and right will give =.

rpnfan, 2022-12-28:
Are their further comments or suggestions regarding the layout?
I suggest to move forward and
a) include the layout in the coming Chrysalis release
b) then I will write the documentation 

I thought about the best language and think German would be more appropriate than English, because the target group are German speaking people only, because qwertz is only used in Germany and Austria.

bomm, 2023-01-07:
The switching from layer 1 (numpad) back to layer 0 does not work as I expected.(both with bodo-5 and beta6)
When I hold `fun` and press `prog`. layer 1 is activated. When I keep `fun`pressed and press `prog` again, layer 1 is deactivated as expected. But when I release `fun`, what I would normally do, and later hold it again, the first press of `prog` seems to activate layer 1, and the second press deactivates it. I would expect to press `fun`+`prog` once to disable layer 1. 
I guess this is because `fun` puts layer 2 on top.
Maybe we have to define `lock to #1` in layer 0 instead of the additional `Esc` key. See attached file.

Is layer 1 supposed to take into account `Num Lock`? It is currently configured to send the normal number keys, not the numpad keys, so it will always send numbers regardless of the `Num Lock` setting. Probably this will not allow to use `Alt`+NumPad to enter special characters.

[qwertz_layout_beta7.json.txt](https://github.com/keyboardio/Chrysalis/files/10366836/qwertz_layout_beta7.json.txt)