keyboardio/Chrysalis #1055: SpaceCadet key doesn't seem to be working

cdwilson, 2022-09-22:
**Describe the bug**
I'm trying to get the SpaceCadet key working.  I assigned it to the prog key, but when I press the SC On key, and then tap the shift keys, nothing happens.

[image]

**Expected behavior**
Tapping the SC On key and then tapping a shift key should cause a `(` or `)` key to be typed.

**Desktop (please complete the following information):**
 - OS: macOS Monterey
 - Chrysalis Version: 0.11.2

davispw, 2022-09-25:
My 2¢: this should be a preference under "My Keyboard" to enable Space Cadet for shift keys permanently, not separate SC On/Off keys.

That said, here's a "me too" report on the latest version: MacOS Ventura Beta; Chrysalis 0.11.3, Keyboardio Model 100 0.11.1+29.  Nothing happens after typing "SC On".

algernon, 2022-09-25:
> My 2¢: this should be a preference under "My Keyboard" to enable Space Cadet for shift keys permanently, not separate SC On/Off keys.

It is, with newer firmware.

pcn, 2022-09-25:
I've tried enabling SC via the prog key, and via the preferences, and neither is working for me. Model 100, firmware version
0.90.1+42 and using Chrysalis version 0.11.3

yet-another-tim, 2022-09-27:
Similar issue for me. Turning space cadet on via Chrysalis has no effect.
Model 100
Ubuntu 20.04
Chrysalis: 0.11.4
Firmware: 0.90.1+42

algernon, 2022-09-27:
I have a fix incoming, it's on the firmware side, so will need a firmware update too. I'll keep this issue open until a suitable firmware bundle is published.

scrthq, 2022-09-29:
Confirming here that firmware 0.90.2 and Chrysalis 0.11.5 has fixed Space Cadet Shift for me on my Model 100 :-) Thanks for the work on this!!!