keyboardio/Chrysalis #1056: Reset EEPROM to factory defaults causes Chrysalis to hang

cdwilson, 2022-09-22:
**Describe the bug**
When I click the "Reset EEPROM to factory defaults", Chrysalis just hangs with the orange spinner:

[image]

After a couple min, I had to kill the app.

**Desktop (please complete the following information):**
 - OS: macOS Monterey
 - Chrysalis Version: 0.11.2

cdwilson, 2022-09-22:
BTW, when I restart Chrysalis, the app connects to the keyboard OK (layout is read), but each time I try to reset the EEPROM, the app hangs.

algernon, 2022-09-22:
Yeah, this is a known problem - but didn't have an issue open about it till now, thanks!.

This happens because Chrysalis wants to wait for the eeprom reset to complete, but for a number of reasons, the firmware reboots the keyboard after resetting eeprom, before sending a reply back to Chrysalis. So Chrysalis keeps waiting for a loooong time.

I have an idea how to fix it, just didn't get around to it yet.

PR #1080 (2022-09-26): When using eeprom.erase, return after 10 seconds
When using `eeprom.erase` to restore factory settings, the keyboard will erase the eeprom and reboot before sending us a reply. We can't await for that to return, as it never will.

While we could - and should - fix the firmware, we still want to handle older firmware, so we need a Chrysalis-side solution too. This solution is  to wait 10 seconds in `clearEEPROM()`, and just return. Ten seconds should be enough to clear the eeprom, even if we're using a pretty large emulated storage where writes are slow. They're not *this* slow.

With this, the [... cut]
commit d4bb6a96 (2022-09-26): When using eeprom.erase, return after 10 seconds

When using `eeprom.erase` to restore factory settings, the keyboard will erase
the eeprom and reboot before sending us a reply. We can't await for that to
return, as it never will.

While we could - and should - fix the firmware, we still want to handle older
firmware, so we need a Chrysalis-side solution too. This solution is  to wait 10
seconds in `clearEEPROM()`, and just return. Ten seconds should be enough to
clear the eeprom, even if we're using a pretty large emulated storage where
writes are slow. They're not *this* slow.

With this, th [... cut]
closed 2022-09-26 by obra