keyboardio/Chrysalis #1059: Importing an existing layout doesn't update colors

fluffy-critter, 2022-09-23:
**Describe the bug**

When importing a layout from a .json file, the key colors do not come through.

**To Reproduce**

1. Export a keyboard layout including colors from a Model 01
2. Import the layout to a Model 100

**Expected behavior**

I expected the key colors to be imported along with the key positions.

**Screenshots**

The importer shows the imported color map:

[image]

But the resulting layout does not include them:

[image]

However, note that the color palette *did* come through. Just not the key assignments.

**Desktop (please complete the following information):**
 - OS: macOS 12.6
 - Chrysalis Version: 0.11.3

The-Compiler, 2022-09-24:
I'm seeing exactly the same. As mentioned in Discord, I was able to transfer my colormap just fine with [focus-send](https://github.com/keyboardio/focus-send) via:

`./focus-send -d /dev/ttyACM0 colormap.map $(./focus-send -d /dev/ttyACM2 colormap.map)`

(with `ttyACM0` being my Model 100, and `ttyACM2` being the Model 01).

Here is the diff of the JSON file after triggering another re-export after that in Chrysalis: https://github.com/The-Compiler/Model01-Firmware/commit/21cca5c2420a4a5483812b9d4ecd1ab1487a5716

Note that for both keys and colormap, there are two entries more, with 64 keys in each. I [set up 6 layers by default](https://github.com/The-Compiler/Model01-Firmware/blob/d51f922f8e5f05a6dae9d090e73c7d9f8699d7b6/Model01-Firmware.ino#L656), while the [Model 100 default](https://github.com/keyboardio/Chrysalis-Firmware-Bundle/blob/9cc0a06c61d5eccb2972e6585dda4f4962349bce/Keyboardio/Model100/Model100.ino#L634) seems to be 8.

So does Chrysalis maybe load keys correctly from a JSON export when the layer count differs, but decides to throw away the LED information?

fluffy-critter, 2022-09-24:
As requested on Discord, here's my model 01 JSON file that failed to import colors on my model 100:

```json
{
  "keymaps": [
    [
      {
        "code": 17453,
        "label": {"hint": "ShiftTo", "base": "#3"},
        "target": 3,
        "rangeStart": 17450,
        "categories": ["layer", "shifttolayer"]
      },
      {"code": 30, "label": {"base": "1", "shifted": "!"}},
      {"code": 31, "label": {"base": "2", "shifted": "@"}},
      {"code": 32, "label": {"base": "3", "shifted": "#"}},
      {"code": 33, "label": {"base": "4", "shifted": "$"}},
      {"code": 34, "label": {"base": "5", "shifted": "%"}},
      {"code": 47, "label": {"base": "[", "shifted": "{"}},
      {
        "code": 225,
        "label": {"base": "Shift"},
        "location": "left",
        "categories": ["modifier"]
      },
      {
        "code": 229,
        "label": {"base": "Shift"},
        "location": "right",
        "categories": ["modifier"]
      },
      {"code": 48, "label": {"base": "]", "shifted": "}"}},
      {"code": 35, "label": {"base": "6", "shifted": "^"}},
      {"code": 36, "label": {"base": "7", "shifted": "&"}},
      {"code": 37, "label": {"base": "8", "shifted": "*"}},
      {"code": 38, "label": {"base": "9", "shifted": "("}},
      {"code": 39, "label": {"base": "0", "shifted": ")"}},
      {"code": 49, "label": {"base": "\\", "shifted": "|"}},
      {"code": 53, "label": {"base": "`", "shifted": "~"}},
      {"code": 20, "label": {"base": "q", "shifted": "Q"}},
      {"code": 26, "label": {"base": "w", "shifted": "W"}},
      {"code": 8, "label": {"base": "e", "shifted": "E"}},
      {"code": 21, "label": {"base": "r", "shifted": "R"}},
      {"code": 23, "label": {"base": "t", "shifted": "T"}},
      {"code": 42, "label": {"base": {"full": "Backspace", "1u": "Bksp"}}},
      {"code": 43, "label": {"base": "Tab"}},
      {"code": 44, "label": {"base": "Space"}},
      {"code": 40, "label": {"base": "Enter"}},
      {"code": 28, "label": {"base": "y", "shifted": "Y"}},
      {"code": 24, "label": {"base": "u", "shifted": "U"}},
      {"code": 12, "label": {"base": "i", "shifted": "I"}},
      {"code": 18, "label": {"base": "o", "shifted": "O"}},
      {"code": 19, "label": {"base": "p", "shifted": "P"}},
      {"code": 46, "label": {"base": "=", "shifted": "+"}},
      {"code": 75, "label": {"base": {"full": "Page Up", "1u": "PgUp"}}},
      {"code": 4, "label": {"base": "a", "shifted": "A"}},
      {"code": 22, "label": {"base": "s", "shifted": "S"}},
      {"code": 7, "label": {"base": "d", "shifted": "D"}},
      {"code": 9, "label": {"base": "f", "shifted": "F"}},
      {"code": 10, "label": {"base": "g", "shifted": "G"}},
      {"code": 41, "label": {"base": "Esc"}},
      {
        "code": 227,
        "label": {"base": {"full": "Command", "1u": "Cmd", "short": "Cm"}},
        "location": "left",
        "categories": ["modifier"]
      },
      {
        "code": 228,
        "label": {"base": {"full": "Control", "1u": "Ctrl"}},
[... 1484 lines cut ...]
      15,
      15,
      15
    ]
  ],
  "palette": [
    {"r": 255, "g": 133, "b": 0, "rgb": "rgb(255, 133, 0)"},
    {"r": 119, "g": 119, "b": 102, "rgb": "rgb(119, 119, 102)"},
    {"r": 85, "g": 136, "b": 170, "rgb": "rgb(85, 136, 170)"},
    {"r": 153, "g": 0, "b": 136, "rgb": "rgb(153, 0, 136)"},
    {"r": 51, "g": 102, "b": 187, "rgb": "rgb(51, 102, 187)"},
    {"r": 127, "g": 136, "b": 0, "rgb": "rgb(127, 136, 0)"},
    {"r": 0, "g": 119, "b": 68, "rgb": "rgb(0, 119, 68)"},
    {"r": 153, "g": 102, "b": 0, "rgb": "rgb(153, 102, 0)"},
    {"r": 85, "g": 85, "b": 119, "rgb": "rgb(85, 85, 119)"},
    {"r": 85, "g": 119, "b": 85, "rgb": "rgb(85, 119, 85)"},
    {"r": 85, "g": 119, "b": 119, "rgb": "rgb(85, 119, 119)"},
    {"r": 136, "g": 136, "b": 204, "rgb": "rgb(136, 136, 204)"},
    {"r": 119, "g": 0, "b": 153, "rgb": "rgb(119, 0, 153)"},
    {"r": 0, "g": 0, "b": 204, "rgb": "rgb(0, 0, 204)"},
    {"r": 119, "g": 51, "b": 85, "rgb": "rgb(119, 51, 85)"},
    {"r": 204, "g": 204, "b": 204, "rgb": "rgb(204, 204, 204)"}
  ]
}
```

The-Compiler, 2022-09-25:
Here is the Model 01 export I'm importing into Chrysalis: [Model01-Firmware/chrysalis.json at custom · The-Compiler/Model01-Firmware](https://github.com/The-Compiler/Model01-Firmware/blob/custom/chrysalis.json)

Here is the Model 100 re-export after transferring the colormap via `focus-send`: [Model01-Firmware/chrysalis.json at model100 · The-Compiler/Model01-Firmware](https://github.com/The-Compiler/Model01-Firmware/blob/model100/chrysalis.json)

(See my comment above for the diff between the two)

Let me note that the colors are just not loaded at all. If I e.g. have this in the file:

[image]

and set a special color for one key:

[image]

after loading the file, it's still that color, and not the black which Chrysalis showed in the preview before clicking "import":

[image]

Here is a debug bundle of loading the file into Chrysalis and then discarding pending changes (thanks, GitHub, for not allowing JSON uploads...): 

[chrysalis-debug.bundle.json.txt](https://github.com/keyboardio/Chrysalis/files/9640269/chrysalis-debug.bundle.json.txt)

The-Compiler, 2022-09-27:
Nevermind what I said above - it's *not* the layer count. When I clear the Model 100's EEPROM via "My keyboard" and then restore the [Model 100 export](https://github.com/The-Compiler/Model01-Firmware/blob/model100/chrysalis.json), my colors are still all set to FF and the colors in the file simply don't get loaded at all.

algernon, 2022-09-27:
Reproduced, the importing from backup & restore is simply borked.

umberleigh, 2022-10-08:
I'm experiencing the same issue.

OS: Windows 10 21H2 build 19044.2006
Chrysalis: 0.11.7
Keyboard: Keyboardio Model 100
Firmware: 0.90.4+49

Snaptags, 2022-10-17:
Any updates on this? A failed firmware update just wiped all colors from my 5 layers :-(

closed 2022-10-23 by obra

Snaptags, 2022-10-24:
@algernon Restoring colors works like a charm using the latest version, thank you very much for fixing this!

PR #1200 (2022-10-23): Fix the colormap importing via backup & restore
When importing the colormap, we're importing both the palette, and the colormap, but we were importing them in two steps, both of which were asynchronously setting the same state. This resulted in the two async state changes stomping on each others feet, and corrupting the colormap.

To remedy this issue, add a new helper that can set them both at the same time, and thus, we avoid the race condition.

Fixes #1059.
commit c277d141 (2022-10-23): Fix the colormap importing via backup & restore

When importing the colormap, we're importing both the palette, and the colormap,
but we were importing them in two steps, both of which were asynchronously
setting the same state. This resulted in the two async state changes stomping on
each others feet, and corrupting the colormap.

To remedy this issue, add a new helper that can set them both at the same time,
and thus, we avoid the race condition.

Fixes #1059.