keyboardio/Chrysalis #1060: Firmware flash fails on Model 100

alarsyo, 2022-09-23:
**Describe the bug**
Firmware flashing fails on the Model 100 on NixOS, using [an updated Chrysalis package](https://git.alarsyo.net/alarsyo/nixos-config/src/commit/791ac1a85c6203ac0a26de4bfba390045300f8fe/overlays/chrysalis/default.nix).

**To Reproduce**
Steps to reproduce the behavior:
1. Launch flashing procedure on a Model 100
2. During step 2, switch to programmable mode, press the PROG key
3. PROG key turns red, then stays that way (stuck in bootloader mode?)
4. See screenshot for what chrysalis displays after a few seconds of the PROG key LED being red

**Expected behavior**
Flashing goes fine

**Screenshots**

[image]

**Desktop (please complete the following information):**
 - OS: NixOS unstable
 - Chrysalis Version: 0.11.3

**Additional context**

The debug bundle logs aren't relevant because when I reproduce the issue, Chrysalis freezes and I can't discard the message window shown in the screenshot. Here's the debug bundle in case there's anything else useful in there, as well as logs grabbed from `.config/chrysalis/logs/` from a time I reproduced the issue:

https://gist.github.com/alarsyo/49386d77aeda45e50232f7d18cb1182b

I do have `/etc/udev/rules.d/60-kaleidoscope.rules` setup correctly, in case that matters.

alarsyo, 2022-09-23:
Here's a debug bundle from flashing a Model 01 firmware just fine: 

<details>

```
{
  "timestamp": "2022-09-23T22:36:39.714Z",
  "uuid": "193a0db9-649e-424c-9de2-27ec1920206e",
  "chrysalis": {"version": "0.11.3"},
  "os": {"platform": "linux", "distro": "NixOS", "release": "22.11", "codename": "Raccoon", "kernel": "5.15.67", "arch": "x64", "hostname": "zephyrus", "fqdn": "zephyrus.alarsyo.net", "codepage": "UTF-8", "logofile": "nixos", "serial": "", "build": "22.11.20220912.5f326e2", "servicepack": "", "uefi": true},
  "usb": [],
  "logs": [
    {"function": "find", "label": "focus", "level": "debug", "message": "serial port list obtained", "portList": [{"manufacturer": "Keyboardio", "path": "/dev/ttyACM0", "pnpId": "usb-Keyboardio_Model_01_Ckbio01E-if00", "productId": "2301", "serialNumber": "Ckbio01E", "vendorId": "1209"}, {"path": "/dev/ttyS0"}, {"path": "/dev/ttyS1"}, {"path": "/dev/ttyS2"}, {"path": "/dev/ttyS3"}], "timestamp": "2022-09-23T22:34:37.047Z"},
    {"devices": [{"focusDeviceDescriptor": {"bootloader": false, "info": {"displayName": "Keyboardio Model 01", "product": "Model01", "urls": [{"name": "Homepage", "url": "https://shop.keyboard.io/"}, {"name": "Forum", "url": "https://community.keyboard.io/"}, {"name": "Chat", "url": "https://keyboard.io/discord-invite"}], "vendor": "Keyboardio"}, "keyboard": {"columns": 16, "rows": 4}, "usb": {"bootloader": {"productId": 8960, "protocol": "avr109", "vendorId": 4617}, "productId": 8961, "vendorId": 4617}}, "manufacturer": "Keyboardio", "path": "/dev/ttyACM0", "pnpId": "usb-Keyboardio_Model_01_Ckbio01E-if00", "productId": "2301", "serialNumber": "Ckbio01E", "vendorId": "1209"}], "function": "find", "label": "focus", "level": "debug", "message": "supported devices found", "timestamp": "2022-09-23T22:34:37.049Z"},
    {"level": "info", "message": "Connecting to port", "path": "/dev/ttyACM0", "timestamp": "2022-09-23T22:34:41.876Z"},
    {"level": "info", "message": "Probing for focus support...", "timestamp": "2022-09-23T22:34:42.003Z"},
    {"label": "focus", "level": "verbose", "message": "request", "request": {"args": [], "command": "help", "id": 0}, "timestamp": "2022-09-23T22:34:42.008Z"},
    {"label": "focus", "level": "verbose", "message": "request", "request": {"args": [], "command": "help", "id": 1}, "timestamp": "2022-09-23T22:34:42.009Z"},
    {"label": "focus", "level": "verbose", "message": "request", "request": {"args": [], "command": "help", "id": 2}, "timestamp": "2022-09-23T22:34:42.009Z"},
    {"label": "focus", "level": "verbose", "message": "request", "request": {"args": [], "command": "help", "id": 3}, "timestamp": "2022-09-23T22:34:42.010Z"},
    {"label": "focus", "level": "verbose", "message": "request", "request": {"args": [], "command": "help", "id": 4}, "timestamp": "2022-09-23T22:34:42.028Z"},
    {"args": ["/editor"], "level": "debug", "message": "navigating to location", "timestamp": "2022-09-23T22:34:42.029Z"},
    {"label": "focus", "level": "verbose", "message": "request", "request": {"args": [], "command": "keymap.default", "id": 5}, "timestamp": "2022-09-23T22:34:42.052Z"},
    {"data": "help\nplugins\r\nsettings.defaultLayer\nsettings.valid?\nsettings.version\nsettings.crc\r\neeprom.contents\neeprom.free\r\n", "label": "focus", "level": "debug", "message": "incoming data", "timestamp": "2022-09-23T22:34:42.266Z"},
    {"label": "focus", "level": "verbose", "message": "response", "request": {"args": [], "command": "help", "id": 0}, "response": "help\nplugins\r\nsettings.defaultLayer\nsettings.valid?\nsettings.version\nsettings.crc\r\neeprom.contents\neeprom.free", "timestamp": "2022-09-23T22:34:42.267Z"},
    {"label": "focus", "level": "verbose", "message": "response", "request": {"args": [], "command": "help", "id": 1}, "response": "help\nplugins\r\nsettings.defaultLayer\nsettings.valid?\nsettings.version\nsettings.crc\r\neeprom.contents\neeprom.free", "timestamp": "2022-09-23T22:34:42.269Z"},
    {"label": "focus", "level": "verbose", "message": "response", "request": {"args": [], "command": "help", "id": 2}, "response": "help\nplugins\r\nsettings.defaultLayer\nsettings.valid?\nsettings.version\nsettings.crc\r\neeprom.contents\neeprom.free", "timestamp": "2022-09-23T22:34:42.274Z"},
    {"label": "focus", "level": "verbose", "message": "response", "request": {"args": [], "command": "help", "id": 3}, "response": "help\nplugins\r\nsettings.defaultLayer\nsettings.valid?\nsettings.version\nsettings.crc\r\neeprom.contents\neeprom.free", "timestamp": "2022-09-23T22:34:42.275Z"},
    {"label": "focus", "level": "verbose", "message": "response", "request": {"args": [], "command": "help", "id": 4}, "response": "help\nplugins\r\nsettings.defaultLayer\nsettings.valid?\nsettings.version\nsettings.crc\r\neeprom.contents\neeprom.free", "timestamp": "2022-09-23T22:34:42.283Z"},
    {"data": ".", "label": "focus", "level": "debug", "message": "incoming data", "timestamp": "2022-09-23T22:34:42.305Z"},
    {"label": "focus", "level": "verbose", "message": "response", "request": {"args": [], "command": "keymap.default", "id": 5}, "timestamp": "2022-09-23T22:34:42.305Z"},
    {"args": [], "command": "keymap.custom", "label": "focus", "level": "verbose", "message": "request (noop)", "timestamp": "2022-09-23T22:34:42.306Z"},
    {"args": [], "command": "keymap.onlyCustom", "label": "focus", "level": "verbose", "message": "request (noop)", "timestamp": "2022-09-23T22:34:42.306Z"},
[... cut ...] dio"}, "keyboard": {"columns": 16, "rows": 4}, "usb": {"bootloader": {"productId": 8960, "protocol": "avr109", "vendorId": 4617}, "productId": 8961, "vendorId": 4617}}, "manufacturer": "Keyboardio", "path": "/dev/ttyACM0", "pnpId": "usb-Keyboardio_Model_01_Ckbio01E-if00", "productId": "2301", "serialNumber": "Ckbio01E", "vendorId": "1209"}], "function": "find", "label": "focus", "level": "debug", "message": "supported devices found", "timestamp": "2022-09-23T22:36:33.949Z"}
  ]
}
```

</details>

alarsyo, 2022-09-23:
I can also add that flashing the latest firmware from a Windows computer worked absolutely fine for the Model 100, so the keyboard itself doesn't seem to be the issue (thankfully :D)

tompi, 2022-09-25:
Im seeing a similar issue at my macs(fails on both intel and m1). But in my setups chrysalis does not even recognise the existing firmware:
[image]

alarsyo, 2022-09-25:
Yeah that's expected, I had the same thing when I first got the keyboard, firmware installed at the factory didn't provide version information. My screenshot above is after I successfully flashed (on windows) a recent firmware (upgraded to 0.90.1 without trouble since then)

UGKEN, 2022-09-26:
I can't flash,too.
Because of same problem.
[image]

TechWilk, 2022-09-26:
Same, I got a flash failed message the first time, then if I hold down PROG while plugging in (so it goes, and stays RED), I get the following debug output from the Console upon trying to flash:

```
2022-09-26T18:40:36.493Z dfu-util exited abnormally 
{
  exitCode: 74
  label: "flash"
  level: "error"
}
log @ transports.js:58

Error while uploading firmware 
{
  error: "dfu-util exited abnormally with an error code of 74"
  level: "error"
}
log @ transports.js:58
```

[image]

tlyu, 2022-09-26:
> Same, and I''m using PopOS and Chrysalis Version: 0.11.4

Did you install the udev rules when prompted? If so, do you have a non-empty file `/etc/udev/rules.d/60-kaleidoscope.rules`?

TechWilk, 2022-09-27:
> Did you install the udev rules when prompted? If so, do you have a non-empty file `/etc/udev/rules.d/60-kaleidoscope.rules`?

No, I wasn't prompted to add any udev rules. Is that something I've missed in the preferences?

algernon, 2022-09-27:
The prompt only appears if you do not have write access to the serial port of the keyboard. If you did, then it's not shown. That's something we'll be fixing in an update soon.

alarsyo, 2022-10-10:
I'm able to flash just fine with the following command, for what it's worth :

```fish
dfu-util --device 3496:0005 -D ~/.config/chrysalis/latest-firmware/Keyboardio/Model100/default.bin -R
```

alarsyo, 2022-10-17:
couldn't get it to work ! still getting similar errors:

[image]

Kilomil, 2022-10-22:
as per request
[chrysalis-debug.json.txt](https://github.com/keyboardio/Chrysalis/files/9845262/chrysalis-debug.json.txt)

TechWilk, 2022-10-24:
Having had this issue (above) on my first M100, I succeeded in flashing it with Windows. However, I've just received my second and tried to flash again on PopOS, receiving the same errors as above.

I get to step 6 of the [forum update guide](https://community.keyboard.io/t/model-100-firmware-update-to-fix-corruption-issues/5553)
- Hold PROG before and while plugging in, to put in program mode. PROG key lights up RED
- Open Chrysalis
- "Connect" to M100
- I am prompted to "Update firmware"
- Choose "latest", with factory reset either disabled or enabled (I've tried both)
- Click ":cloud: Update"
- Click continue to confirm
- ERROR, I am taken back to the homescreen with an "ERROR" toast notification

In the console I can see
```
dfu-util exited abnormally 
    {exitCode: 74, level: 'error', label: 'flash'}
    exitCode: 74
    label: "flash"
    level: "error"
Error while uploading firmware 
    {error: 'dfu-util exited abnormally with an error code of 74', level: 'error'}
    error: "dfu-util exited abnormally with an error code of 74"level: "error"
focus.js:344 Uncaught (in promise) TypeError: Cannot read properties of null (reading 'bootloader')
    at focus_Focus.isInApplicationMode (webpack:/src/api/focus.js:344:39)
    at focus_Focus.request (webpack:/src/api/focus.js:503:15)
    at focus_Focus.supported_commands (webpack:/src/api/focus.js:483:45)
    at new FocusCommands (webpack:/src/api/flash/FocusCommands.js:27:17)
    at flash (webpack:/src/api/flash.js:48:25)
    at Object.flash (webpack:/src/api/hardware-keyboardio-model01/index.js:109:12)
    at _flash (webpack:/src/renderer/screens/FirmwareUpdate.js:104:34)
    at upload (webpack:/src/renderer/screens/FirmwareUpdate.js:151:13)
```
I've tried both Chrysalis 0.11.9 and 0.12.0-snapshot.189
[chrysalis-debug-12snapshot.json.txt](https://github.com/keyboardio/Chrysalis/files/9854856/chrysalis-debug-12snapshot.json.txt)
[chrysalis-debug-11-9.json.txt](https://github.com/keyboardio/Chrysalis/files/9854857/chrysalis-debug-11-9.json.txt)

Let me know if there's anything further I can do to help debug this while I still have a keyboard on the factory firmware.

TechWilk, 2022-10-24:
> This sounds like a case of #1068, Chrysalis not prompting to add udev rules when it should. Can you add the udev rules from [here](https://raw.githubusercontent.com/keyboardio/Chrysalis/master/static/udev/60-kaleidoscope.rules), and see if flashing still fails?

Thank you for your prompt reply. Yes I have now added those rules into `/etc/udev/rules.d/60-kaleidoscope.rules`, rebooted and flashing succeeded!

pgallaway, 2022-10-25:
Had posted issues on the community portal at keyboard.io. Firmware upgrade was not restoring eeprom contents and then not restoring saved color configuration. Layers were not maintained (and in one case totally different). Flashed latest firmware (0.91.0-snapshot-64) with Chrysalis 0.11.19 (release) and layers and color assignments went smoothly. Bundle attached.
[chrysalis-debug20221025.json.txt](https://github.com/keyboardio/Chrysalis/files/9864162/chrysalis-debug20221025.json.txt)

obra, 2024-02-26:
Chrysalis now runs in-browser using WebSerial and WebUSB and this functionality has been rewritten, so I'm closing out this issue as obsolete. Please don't hesitate to open a new issue if https://chrysalis.keyboard.io exhibits the same behavior

closed 2024-02-26 by obra

jwillikers, 2025-01-23:
@obra I just hit this issue attempting to update my Model 100 using Chrysalis from the browser. Nothing prompted me about udev rules, even though I didn't have any ones for my keyboard installed. Seems like maybe it's supposed to? I had to add the ones mentioned in https://github.com/keyboardio/Chrysalis/issues/1060#issuecomment-1289530821. Would have helped if these were referenced somewhere in the README, the Wiki, or Chrysalis.

obra, 2025-01-23:
> mentioned in #1060 (comment)

What happened when it failed to connect? It should have shown you an error
page?

jwillikers, 2025-01-24:
> What happened when it failed to connect? It should have shown you an error
> page?
> […](#)

It hangs on the Firmware Update page during the `flashing` step. The keyboard successfully enters bootloader but stays at the flashing step forever.