keyboardio/Chrysalis #1062: selected key outline should be more visible

tlyu, 2022-09-24:
**Describe the bug**

The outline that highlights the selected key is very thin, and doesn't always contrast adequately with the canvas background or the fill color of the key. This might be worse on Retina or other HiDPI displays.

**To Reproduce**
1. Start Chrysalis.
2. Select the keyboard and click "CONNECT".
3. This brings up the "Layout & Colormap Editor" screen.
4. Observe that the outline of the "prog" key is very faint, especially with a light background. (I think "prog" is the key that's selected by default when launching the layout editor.)

**Expected behavior**
The outline that highlights the selected key should be plainly visible, even on Retina or HiDPI displays, regardless of the key fill color or the background color of the canvas. At the very least, it should be heavier in weight. This might also require something like multiple nested outlines to ensure contrast, or adaptively changing the color of the outline based on its surroundings.

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: macOS 10.15.7 (19H2026)
 - Chrysalis Version: Chrysalis 0.11.3

**Additional context**
N/A

tlyu, 2022-09-25:
Experimented with manually editing the DOM in Developer Tools. It looks like setting `stroke-width="4"` and making the stroke color more saturated helps a lot. I'm not sure why these attributes are hardcoded in the SVG elements instead of being set via CSS.

algernon, 2022-09-25:
To answer the "why" part: because algernon is bad at building UIs, and he didn't know better when he wrote that part of the code some years ago. O:) (sorry)

Setting it via CSS is definitely the way forward.

tlyu, 2022-09-28:
#1098 is an improvement, but for some reason the outlines on the palm keys remain thin. I'm experimenting with a possible fix. I think it's because of the different transform being used for the ellipses.

PR #1109 (2022-09-28): improve key outlines
Make palm key stroke width wider to compensate for the different tranformation used for these shapes. Also slightly increase the stroke width for other keys.

Fixes #1062.
closed 2022-09-28 by obra