keyboardio/Chrysalis #1064: unable to find documentation

tlyu, 2022-09-24:
**Describe the bug**
It's not obvious where to find documentation for Chrysalis.

**To Reproduce**
Steps to reproduce the behavior:
1. Start Chrysalis.
2. Attempt to locate any sort of in-app help or documentation.
3. Observe that the only menu items with an "info" item are the "Layout Card" and "Report a problem" items.
4. Attempt to locate help on the web. Observe that the best pointers are to the GitHub repository, which has only limited documentation, and has more information about installing and developing Chrysalis than about how to use it.

**Expected behavior**
There should be multiple obvious pointers to documentation for Chrysalis, and possibly in-app help of some sort. Existing documentation on the GitHub repository (or elsewhere) should give more help with actually using Chrysalis.

The colormap capabilities seem especially confusing without a high-level overview of how they're supposed to function. It's possible to piece together some of this knowledge from the plugin documentations, but this is not very friendly to end-users.

**Screenshots**
N/A

**Desktop (please complete the following information):**
 - OS: 10.15.7 (19H2026)
 - Chrysalis Version: Chrysalis 0.11.3

**Additional context**
N/A

nebkor, 2022-09-24:
As an example, I just now learned the way to print a layout card (`ctrl-p`), from seeing a bug where someone tried that and it didn't work.

algernon, 2022-09-25:
For what it's worth, there's a [Tour of Chrysalis](https://github.com/keyboardio/Chrysalis/wiki/A-Tour-of-Chrysalis) page on the wiki - although it isn't trivial to find. Mostly because it gets outdated quickly, and I do not have the resources to keep it updated all the time - too many moving parts.

Ideally, we'd have an in-app tour (and a help system), but that has prerequisites which we do not have yet.