keyboardio/Chrysalis #1066: Make it easier to set the color on lots of keys at one time

bcparkison, 2022-09-25:
**Is your feature request related to a problem? Please describe.**

When first configuring a new keyboard, I have to set the color for a lot of keys.  It's time consuming, and is made even more so when bugs cause the color settings to be lost.

**Describe the solution you'd like**

I think a couple of different options would be useful:
* Set the color for all keys defined on a particular layer at one time
* Allow selecting multiple keys and setting the color for all of them at once

**Describe alternatives you've considered**

I haven't considered any other alternatives, but am certainly open to the idea better solutions exist!

tlyu, 2022-09-25:
Another related suggestion: add the ability to fill an entire layer with a given keycode. (I'm guessing the most common use case will be Blocked or Transparent.)

fc-amiggi, 2022-09-28:
I'm interested in this feature. Adding a note here to track.