keyboardio/Chrysalis #1067: colormap not backed up and restored

bcparkison, 2022-09-25:
**Describe the bug**

While working my way through custom firmware and corruption bugs, I backed up and restored my keymap a lot of times, in hopes of not having to recreate it after each corruption.  The keymap itself seemed to do the backup-restore round trip just fine, but I could never get the colormap to restore correctly.  It remembers the colors I have defined, but seemed to mostly scatter them around randomly.

**To Reproduce**

1. Create a keymap and add some colors to some of the keys
2. Backup the keymap to a file
3. I always did lots of things in here, up to, and including, flashing new firmwares.  I don't know how much of this is required.
4. Restore the keymap from the previously created json file
5. See that the keymap is fine, but the colormap is not

**Expected behavior**

The colormap is also correctly restored.

**Desktop (please complete the following information):**
 - OS: Linux (EndeavourOS, if it matters)
 - Chrysalis Version: 0.11.3
 - Kaleidoscope Version: 0.90.0, 0.90.1, custom based on 0.90.1 (not sure if this matters)

**Additional context**

I honestly didn't try again after I got the corruption fix into my custom firmware, because if I successfully repro the problem, I have to manually recreate the colomap again, and I already did that so many times, I couldn't bear the thought of doing it again!

tlyu, 2022-09-26:
I'm not able to replicate this on a Model 100 with 0.90.1-based firmware and Chrysalis 0.11.4. I tried assigning color 14 (edited to orange) to the "fun" keys on layer 2, and that was retained across a layout backup and restore.

When you were getting colormap corruption, which layers were getting their color assignments scrambled? Did it include Layer 0, or only the higher layers?

How exactly were you building your custom firmware "based on 0.90.1"? Were you using a copy of Kaleidoscope checked out from GitHub, with the GD32 BSP checkout also updated to include the USB fix?

bcparkison, 2022-09-26:
Ok, using the latest Chrysalis and a firmware known to contain the corruption fix, I'm not reproing this problem.  I'm going to close this out.  Sorry for the pilot error!

closed 2022-09-26 by bcparkison

maralorn, 2022-10-11:
Hey, I think I am encountering exactly this bug.

OS: Linux (NixOSR
Chrysalis Version: 0.11.5 & 0.11.8
Kaleidoscope Version: 0.90.6

At, first updating the firmware from 0.90.3 to 0.90.6 completely garbled my keys and colors. (Did that with 0.11.8.)
Then I succesfully restored the keys from a backuped json, but everytime I load the backup the colors get mixed up like described here.

Let me demonstrate:

1. Go to Layout -> Restore & Backup -> Load from File
2. See this: 
[image]
3. Click import, confirm.
4. See this:
[image]

If it is not clear, the colors displayed in the first image are the way I configured them.

I am suspecting this to be a chrysalis bug because the colors get garbled between loading and importing and before writing to the keyboard.