keyboardio/Chrysalis #1070: Fun key stops working

GreenerIce, 2022-09-25:
Steps I used to replicate (on Arch Linux) the issue for the Model 100

- Load Chrysalis
- Reset to factory settings - everything working great
- Remap "Prog" to "Super + x" and "Any" to "Super + d"
- "Fun" key stops working (does nothing)

Repeating the above steps a second time (or third time) gives me exactly the same results.

algernon, 2022-09-25:
Quick question I forgot to ask on Discord: what version of Chrysalis and firmware are you on? If you're not on 0.11.3 & 0.90.1 respectively, can you try reproducing the problem with those?

GreenerIce, 2022-09-25:
Firmware is 0.11.1+29 (latest version, according to the page), Chrysalis is 0.11.3

v0.90.1 sounds rather different?

algernon, 2022-09-25:
Yes, 0.90.1 is the newest. You can enable automatic firmware updates in Preferences, and Chrysalis will pull down 0.90.1, which you can then flash.

The reason why I asked is because there was a major firmware bug in all prior versions, and this issue sounds exactly like something that would cause. [More information here](https://community.keyboard.io/t/important-firmware-release-for-the-model-100/5553).

GreenerIce, 2022-09-25:
Ah, yes!  Setting the option for automatic updates seemed to do the trick, thanks!  Had to flash a couple of times - the first time it looked like it worked okay, although I got a warning message in Chrysalis to say it had failed... second time I didn't get the warning message.  Now I can successfully save an updated layout and still have fun (the key working).

closed 2022-09-25 by GreenerIce