keyboardio/Chrysalis #1077: Cannot connect to Atreus using Chrysalis

ksheng, 2022-09-25:
*Describe the bug*
When using Chrysalis to connect to an Atreus keyboard, instead of displaying the keymap, a blank screen is displayed.

Errors in the development console are below:

[image]

*To Reproduce:*
* Open Chrysalis at version 0.11.3
* Plug in the Atreus
* Select it in the Chrysalis UI and click "Connect"

*Expected behavior:*
The keymap view is displayed when connecting.

*Desktop (please complete the following information):*
* OS: MacOS Monterey 12.6
* Chrysalis Version: 0.11.3

obra, 2022-09-26:
Hi Kevin,

Is this the first time you're using Chrysalis with this Atreus or is this a
regression for you? Could you give 0.11.4 a shot to see if it behaves
better?

ksheng, 2022-09-26:
Hey Jesse, this is a regression from a very old version. The Atreus had been working normally with (I believe) version 0.7.9 from July 2020.

I just tried with 0.11.4 and it was still the same behavior.

obra, 2022-09-26:
Can you go into the Chrysalis prefs and turn on 'verbose logging':

[image]

And then try the connection. When it fails, head over into the menu and pick "Report a problem" and download and attach the debugging bundle .json.txt file it generates?

ksheng, 2022-09-27:
Ah, ran into a snag. I enabled verbose logging, but when I go to click "Connect", the entire window goes white (the buggy behavior) and there is no longer a "Report a problem" button.

Where does the app write to in the filesystem? I can try finding it from the command line.

obra, 2022-09-27:
If you open up the developer console, (the radio button right next to
verbose logging), then you should get the bits we really need in the
console (which won't go white)

ksheng, 2022-09-27:
That worked! Here's the downloaded log from the developer console:

[atreus.log](https://github.com/keyboardio/Chrysalis/files/9651356/atreus.log)

bjarkevad, 2022-10-03:
I'm seeing that same issue with Chrysalis 0.11.5 on Linux (Arch Linux, using AppImage installed from here https://aur.archlinux.org/packages/chrysalis-bin) and two different Atreus keyboards. Connecting to a Model 100 works correctly.

bjarkevad, 2022-10-09:
[image]
I'm seeing this error with the latest snapshot. I've flashed a few simple macros on both keyboards back when I first got them, could that be causing problems? The "Report a problem" option is causing this build to freeze on my laptop, but I took this screenshot of the console 
[image]

bjarkevad, 2022-11-04:
Chrysalis works after flashing stock Kaleidoscope!

ksheng, 2022-11-08:
On the latest version (Chrysalis 0.12.0), I'm now getting this error instead of a blank screen when trying to connect. Here's the error message, and I've attached the [debug log](https://github.com/keyboardio/Chrysalis/files/9965293/atreus.log) from the developer console as well.

[image]

obra, 2024-02-26:
Chrysalis now runs in-browser using WebSerial and WebUSB and this functionality has been rewritten, so I'm closing out this issue as obsolete. Please don't hesitate to open a new issue if https://chrysalis.keyboard.io exhibits the same behavior

closed 2024-02-26 by obra