keyboardio/Chrysalis #1083: Save dynamic macros (and config?) to JSON exports

The-Compiler, 2022-09-27:
Today I cleared my keyboard's EEPROM to debug something, and was surprised that my macros were gone when restoring the backup. I'm (mostly) okay with the settings under "my keyboard" being gone (those are easy to set up again), but I'd have expected a backup to also include and restore the dynamic macros.

algernon, 2022-09-27:
FWIW, the macros *are* saved. It's the restore that's not functioning properly. So once I fix the restore, your previous backups will restore your macros too.

The-Compiler, 2022-09-27:
That's not what I'm seeing, unfortunately. After configuring dynamic macros, saving them to the keyboard, and doing a re-export, here is the diff I get: https://github.com/The-Compiler/Model01-Firmware/commit/84fcca65595b0c20f06a38e4f54f57771928f13e

algernon, 2022-09-27:
Ooof, indeed, my bad. I was looking at a backup json made during flashing (which does include pretty much everything). The export from the backup & restore screen does not.