keyboardio/Chrysalis #1084: Tried to change one key, ended up losing all the other keys

HMMcIntosh, 2022-09-27:
**Describe the bug**
I changed one key and downloaded.  Now the key I changed is the *only* one on the keyboard that works.  How do I get my keyboard back?

**To Reproduce**
Steps to reproduce the behavior:
1C lick on a key.
2 Choose the key you want it to be.
3 Click the download button.

**Expected behavior**
I expected just the one key I changed to be altered.  Instead all my other keys were disabled.

**Screenshots**
If applicable, add screenshots to help explain your problem. If possible, please also include the debug bundle you can create on the "Report a problem" screen. If, for some reason, you are unable to access that screen, please press `Control + Shift + I` (Windows or Linux) or `Command + Option + I` (macOS) to bring up the developer console. Right-clicking inside it will allow you to save its contents to a file - please attach that to the bug report.

**Desktop (please complete the following information):**
 - OS: Windows 10
 - Chrysalis Version: 0.11.4

**Additional context**

tlyu, 2022-09-27:
Does this occur after installing fully upgraded firmware (0.90.1) and doing a factory reset? Instructions are in https://community.keyboard.io/t/model-100-firmware-update-to-fix-corruption-issues/5553

HMMcIntosh, 2022-09-27:
Yes, I went through the broken and murky process of upgrading the firmware.  I just found the hidden and buggy process of doing a factory reset, and that fixed the problem. 

I will still probably throw my Keyboardio away.  Not because it's not a great keyboard—it is great—but because I was promised software to let me customize it, and the only software available is Chrysalis, which is beyond useless.  If I can't customize the keyboard, I don't want it.

Harry McIntosh
Software Developer
PAC Software, Inc. [EmailLogo.gif]

HMMcIntosh, 2022-09-27:
> Chrysalis is not the only software. The firmware is equally open source, and lets you customise your keyboard even more. If you're unhappy with Chrysalis, you have other options. The protocol between the keyboard and Chrysalis is also open, so you can even write your own tool, would you so desire.
> For reference, there's a command line tool in Rust [https://github.com/keyboardio/focus-send] (or the same thing in shell [https://github.com/keyboardio/Kaleidoscope/blob/master/bin/focus-send]), an emacs package [https://git.madhouse-project.org/algernon/kaleidoscope.el] that can also talk to the keyboard. Their interface is considerably lower level than that of Chrysalis', but you can script your keymap updates with them nevertheless.

I don't think the fact that everything is open source is a solution.  I have neither the time nor the interest in writing my own tool.  I don't even have the time to spend learning how to use a low-level tool.

The Keyboardio people should be embarrassed by the quality of Chrysalis, and should be providing a good GUI tool as an alternative.  

Harry McIntosh
Software Developer
PAC Software, Inc. [EmailLogo.gif]

obra, 2022-09-27:
Hi Harry. 

You sound pretty frustrated. I would be too if I'd had the experience it sounds like you did while trying to set up your keyboard.

I'm really sorry Chrysalis hasn't treated you well and really appreciate that you took the time to file issues against it.  If you end up deciding that you'd like to try customizing your layout again, I'd be very happy to work through the process with you.

Going forward, I'd greatly appreciate it if we could keep the tone of discussion in public github issues constructive.  We're actively trying to help and it's hard to do that when you tell us that you plan to throw your keyboard away and that we "ought to be embarrassed" for shipping software that is "beyond useless."

(I'm closing this issue because it sounds like the specific problem you described was solved, albeit inelegantly, with a firmware upgrade and EEPROM reset.)

Best,
Jesse

closed 2022-09-27 by obra

HMMcIntosh, 2022-09-28:
I'm sorry if my tone got out of hand.  I'll try to improve it.

But I have to say that I think you've missed the real problem here.  The problem you seem to be dealing with is that I didn't know how to operate the software, and now that I've managed to do one thing successfully (restore my keyboard layout which Chrysalis destroyed), then everything is OK.

It's not OK.  I ordered this keyboard a long time ago, and I've been patiently waiting for it.  The hardware is exceptionally good, but it was supposed to come with software that let me change the layout.  The software it came with is of very poor quality.  You would never ship hardware like that; why do you let software of that quality out?  The software is full of bugs, it's very hard to operate, and if you look at the version number it's clearly not ready for release yet (it starts with "0.").  

What are you going to do to fulfill your committment to provide good quality software to your customers?

Harry McIntosh
Software Developer
PAC Software, Inc. [EmailLogo.gif]