keyboardio/Chrysalis #1086: Setting layer move using Chrysalis does not handle transparent keys correctly

ksheng, 2022-09-27:
**Describe the bug**
When using Chrysalis to customize keymaps, the Move To Layer functionality behaves like Lock to Layer as described in the [Kaleidoscope documentation](https://kaleidoscope.readthedocs.io/en/latest/layers.html#layer-keys).

**To Reproduce**
Steps to reproduce the behavior:
1. Start with layer 0 having the default keymap.
2. Put a transparent key on layer 1 over the physical "A" key.
3. Configure a key to either be "Move to Layer 1" or have it be a Qukey with "Shift to Layer 1 when held" (both cases produce the bug). 
4. While using the configured key to activate layer 1, press the physical "A" key.

**Expected behavior**
Nothing happens.

**Observed behavior**
The letter "a" is sent.

**Desktop (please complete the following information):**
 - OS: MacOS Monterey 12.6
 - Chrysalis Version: 0.11.4

**Additional Context**
The expected behavior is seen by setting "Blocked" instead of "Transparent" on layers where I want to disable the key.

algernon, 2022-09-27:
This is a shortcoming / bug in the firmware: it treats layer 0 as always active. It probably shouldn't do that.

ksheng, 2022-09-27:
Ah okay, I see, I can continue to used "Blocked" as a workaround then.

closed 2024-11-11 by obra