keyboardio/Chrysalis #1087: Layout editor UX is confusing

tlyu, 2022-09-27:
**Describe the bug**
Multiple aspects of the sidebar in the layout editor are confusing. It was difficult for me to figure out how to use the layout editor.

**To Reproduce**
N/A; these are overall impressions based on a large number of interactions with the layout editor.

**Expected behavior**
The layer summary for the selected key should have its second column read something like "Key code/action" instead of "Key #N". There could be a heading that reads something like "Key #N summary" above that summary table.

There should be a heading above the per-layer settings that reads something like "Layer NAME Key #N", to make it clear that the settings below apply to a specific key and layer. Maybe add some explanatory text like "To assign a standard keyboard key code, click on the key in the key picker." Possibly make this heading sticky to the top of the sidebar when the sidebar is scrolled.

The color picker should have only the palette squares, ideally laid out in a more symmetrical configuration (2 rows of 8, maybe). There should be a separate button to open a palette editor, and picking palette entries in the palette editor should _not_ change the assigned color of the selected key.

**Screenshots**
N/A

**Desktop (please complete the following information):**
N/A

**Additional context**
N/A

obra, 2022-09-27:
I'm going to break this apart into a few tickets. 

It's 100% ok for bug reports to contain multiple issues, but to be resolvable, "work" tickets should be one-per-issue.

obra, 2022-09-27:
Replaced by #1090, #1091, #1092

tlyu, 2022-09-27:
> I'm going to break this apart into a few tickets.
> 
> It's 100% ok for bug reports to contain multiple issues, but to be resolvable, "work" tickets should be one-per-issue.

That's fair. I think the color picker change is easily separable. I think the key/layer sidebar heading changes should probably happen together.

closed 2022-09-27 by obra