keyboardio/Chrysalis #1088: Possible corruption with LayerNames

algernon, 2022-09-27:
We've been seeing reports of corrupted data being saved, even with Chrysalis 0.11.4 and 0.90.1 firmware. They seem to be related to editable layer names.

obra, 2022-09-27:
One description of the issue: https://canary.discord.com/channels/492408953041321984/761681781447196693/1024153691978223656

tlyu, 2022-09-27:
On a Model 100 with fully updated firmware on macOS, I still can't replicate the layer name corruption, but there seems to be something strange going on with how the firmware stores layer names, and that might be related:

```
keymap.layerNames
19 Test123456789012345
8 Multiple
22 Testing something else
130 Here is a ridiculously long layer name to see what breaks if we make something this horribly lengthy and verbose =asz*)!_z\z!@#$%&
0 size=136
.
```
(The punctuation characters in the layer name are intentional.)

I did create a custom layer name for layer 4, but that seems to not be stored. Note the layer name lengths total greater than the reserved 136 bytes. After I wrote the long layer names to the keyboard, Chrysalis still continued to show my custom layer name for layer 4, until I quit and restarted Chrysalis. (I quit Chrysalis so that I could query the layer names interactively.)

algernon, 2022-09-27:
I think I have a hunch about the corruption, but it requires the layernames to overflow: in that case, the plugin will simply write into the area of another plugin. In the default firmware, that's DefaultLEDModeConfig, and then some unallocated area. In custom firmware, if the LayerNames plugin comes earlier, it can overwrite whatever comes after.

The reason for the possible overflow is this part:

https://github.com/keyboardio/Kaleidoscope/blob/28d6d66e98f0d85170191c76bff8e970b92081f4/plugins/Kaleidoscope-LayerNames/src/kaleidoscope/plugin/LayerNames.cpp#L74-L79

We should have a bounds check in that for loop. Or better yet, check if we'd go out of bounds *before* storing the length in the first place.

algernon, 2022-09-27:
FWIW, I applied the same layer names as above, using the `focus-send`  tool, and Chrysalis displays all of them (using the stock 0.90.1 firmware). I did need to click the "show empty layers" button to show the 4th, but once I did, it was there, and clicking into the editable box, I could jump to the end, and saw the punctuation characters at the end.

tlyu, 2022-09-27:
> FWIW, I applied the same layer names as above, using the `focus-send` tool, and Chrysalis displays all of them (using the stock 0.90.1 firmware). I did need to click the "show empty layers" button [...]

Right. Not shown in that dump is a fifth layer name (number 4) that ended in a long string of digits. That one was not saved by Focus, for one reason or other.

algernon, 2022-09-27:
Ah. That one wasn't saved, because the plugin's outer loop discarded it:

https://github.com/keyboardio/Kaleidoscope/blob/28d6d66e98f0d85170191c76bff8e970b92081f4/plugins/Kaleidoscope-LayerNames/src/kaleidoscope/plugin/LayerNames.cpp#L60

Chrysalis should warn about such overflows, like it does when saving Macros,  to avoid surprises like this. (And the firmware should be more careful about not overflowing, too)

tlyu, 2022-09-27:
It's also worth noting that the user in that Discord thread reported ongoing communication timeouts on macOS after upgrading to firmware 0.90.1. There may be unidentified unrelated serial communication factors causing this corruption, rather than Chrysalis or firmware bugs.

algernon, 2022-09-27:
This has largely been addressed by #1100, at least a big chunk of it. Keeping this open until keyboardio/Kaleidoscope#1247 gets addressed too, and trickles down to shipped firmware, and until we can verify there's no other shenanigans going on.

tlyu, 2022-09-27:
I don't know whether my observed bounds-checking problem is the cause of what the original user reported. They claimed that their largest custom layer name was 8 characters. However, I would totally believe that another communication problem could have resulted in an excessively long custom layer name, with subsequent compounding corruption.

algernon, 2022-09-27:
Perhaps a previous layer name corruption that persisted? Though, I don't see how messed up layer names would trickle into keymaps, because the storage space for keymaps and colors are allocated before layernames in the EEPROM layout. So unless we wrap around somehow, I don't see how layernames would be the culprit.

At this point, I'd be suspecting something else, too.

obra, 2024-11-11:
I believe this to now be resolved. Please feel free to reopen if you see it flare up again

closed 2024-11-11 by obra