keyboardio/Chrysalis #1093: MacOS sleep persistent enumeration failures

richarddli, 2022-09-27:
**Describe the bug**

When I put my Mac to sleep, it does not actually sleep, and resumes wake.

**To Reproduce**
Steps to reproduce the behavior:
1. Put Mac to sleep.
2. Discover it does not sleep.

**Screenshots**
```
[116702.983870]: 116702.983868 AppleUSB20XHCIPort@01400000: AppleUSBHostPort::disconnect: persistent enumeration failures
[116703.057964]:  12 b 20MHz n CCA 6% duration: 112 congest: 7 ts: 116708
[116703.097617]:  36 a 20MHz n CCA 3% duration: 774 congest: 26 ts: 116708
[116703.215258]:  13 b 20MHz n CCA 7% duration: 113 congest: 8 ts: 116708
[116703.279294]:  36 a 20MHz n CCA 3% duration: 832 congest: 29 ts: 116708
[116703.395625]:  52 a 20MHz n CCA 0% duration: 112 congest: 0 ts: 116708
[116703.460996]:  36 a 20MHz n CCA 3% duration: 371 congest: 12 ts: 116708
[116703.577773]:  56 a 20MHz n CCA 0% duration: 113 congest: 0 ts: 116708
[116703.616383]:  36 a 20MHz n CCA 3% duration: 404 congest: 13 ts: 116708
[116703.632501]: 116703.632500 AppleUSB20XHCIPort@01400000: AppleUSBHostPort::disconnect: persistent enumeration failures
[116703.732261]:  60 a 20MHz n CCA 0% duration: 112 congest: 0 ts: 116708
[116703.796825]:  36 a 20MHz n CCA 3% duration: 462 congest: 14 ts: 116708
[116703.913278]:  64 a 20MHz n CCA 0% duration: 113 congest: 0 ts: 116708
[116703.978613]:  36 a 20MHz n CCA 3% duration: 522 congest: 16 ts: 116708
[116703.996530]: LQM-WiFi:TX(D2:21:F9:9F:14:D2) AC<SU MS NB NRS NA CM EX TF FFP MRET FLE> BE<35 0 0 0 0 0 0 0 0 0 0> (5000ms)
[116703.996541]: LQM-WiFi:TX(D2:21:F9:9F:14:D2) AC<SU MS NB NRS NA CM EX TF FFP MRET FLE> BK<0 0 0 0 0 0 0 0 0 0 0> (5000ms)
[116703.996549]: LQM-WiFi:TX(D2:21:F9:9F:14:D2) AC<SU MS NB NRS NA CM EX TF FFP MRET FLE> VO<0 0 0 0 0 0 0 0 0 0 0> (5000ms)
[116703.996560]: LQM-WiFi:L3 Control VO TX(D2:21:F9:9F:14:D2) Success=0 NoACK=0 Expired=0 OtherErr=0
[116703.996564]: LQM-WiFi:TX(FF:FF:FF:FF:FF:FF) AC<SU MS NB NRS NA CM EX TF FFP MRET FLE> BE<0 0 0 0 0 0 0 0 0 0 0> (5000ms)
[116704.039265]: [IGFB][LOG  ][DISPLAY   ] ProAudioTest: proAudioWorkLoad notification timestamp = 116704039225113, state = Disable Wake
[116704.095851]: 100 a 20MHz n CCA 0% duration: 113 congest: 0 ts: 116709
[116704.102565]: filecache_entry_invalidate: invalidating <ptr> 36 a 20MHz n CCA 3% duration: 555 congest: 17 ts: 116709
[116704.139062]: tx_flush:1182: disk1 xid 24945875 tx stats: # 98680 finish 92172 enter 419 wait 9919 1791us close 48us flush 1616us
[116704.251076]: 104 a 20MHz n CCA 0% duration: 112 congest: 0 ts: 116709
[116704.283418]: 116704.283415 AppleUSB20XHCIPort@01400000: AppleUSBHostPort::disconnect: persistent enumeration failures
[116704.315398]:  36 a 20MHz n CCA 2% duration: 614 congest: 18 ts: 116709
[116704.432525]: 108 a 20MHz n CCA 0% duration: 113 congest: 0 ts: 116709
[116704.497533]:  36 a 20MHz n CCA 3% duration: 289 congest: 10 ts: 116709
[116704.614737]: 112 a 20MHz n CCA 0% duration: 113 congest: 0 ts: 116709
[116704.639245]: [IGFB][LOG  ][DISPLAY   ] ProAudioTest: proAudioWorkLoad notification timestamp = 116704639162228, state = Enable wake
[116704.655615]:  36 a 20MHz n CCA 3% duration: 323 congest: 10 ts: 116709
[116704.772516]: 116 a 20MHz n CCA 0% duration: 113 congest: 0 ts: 116709
[116704.836964]:  36 a 20MHz n CCA 2% duration: 382 congest: 10 ts: 116709
[116704.933455]: 116704.933449 AppleUSB20XHCIPort@01400000: AppleUSBHostPort::disconnect: persistent enumeration failures
[116704.954063]: 120 a 20MHz n CCA 0% duration: 113 congest: 0 ts: 116709
[116705.018106]:  36 a 20MHz n CCA 3% duration: 440 congest: 15 ts: 116709
[116705.134582]: 124 a 20MHz n CCA 0% duration: 112 congest: 0 ts: 116710
[116705.175552]:  36 a 20MHz n CCA 3% duration: 475 congest: 15 ts: 116710
[116705.292543]: 128 a 20MHz n CCA 0% duration: 113 congest: 0 ts: 116710
[116705.357700]:  36 a 20MHz n CCA 2% duration: 534 congest: 16 ts: 116710
[116705.474401]: 132 a 20MHz n CCA 0% duration: 113 congest: 0 ts: 116710
[116705.539585]:  36 a 20MHz n CCA 3% duration: 259 congest: 9 ts: 116710
[116705.583665]: 116705.583660 AppleUSB20XHCIPort@01400000: AppleUSBHostPort::disconnect: persistent enumeration failures
[116705.656369]: 136 a 20MHz n CCA 0% duration: 113 congest: 0 ts: 116710
[116705.695850]:  36 a 20MHz n CCA 3% duration: 293 congest: 10 ts: 116710
[116705.812808]: 140 a 20MHz n CCA 0% duration: 113 congest: 0 ts: 116710
[116705.877113]:  36 a 20MHz n CCA 3% duration: 351 congest: 11 ts: 116710
[116705.993133]: 144 a 20MHz n CCA 0% duration: 112 congest: 0 ts: 116710
[116706.039744]: [IGFB][LOG  ][DISPLAY   ] ProAudioTest: proAudioWorkLoad notification timestamp = 116706039728187, state = Disable Wake
[116706.232869]: 116706.232864 AppleUSB20XHCIPort@01400000: AppleUSBHostPort::disconnect: persistent enumeration failures
[116706.639477]: [IGFB][LOG  ][DISPLAY   ] ProAudioTest: proAudioWorkLoad notification timestamp = 116706639461968, state = Enable wake
[116706.884162]: 116706.884157 AppleUSB20XHCIPort@01400000: AppleUSBHostPort::disconnect: persistent enumeration failures
[116707.532583]: 116707.532582 AppleUSB20XHCIPort@01400000: AppleUSBHostPort::disconnect: persistent enumeration failures
[116708.039610]: [IGFB][LOG  ][DISPLAY   ] ProAudioTest: proAudioWorkLoad notification timestamp = 116708039519845, state = Disable Wake
[116708.181673]: 116708.181671 AppleUSB20XHCIPort@01400000: AppleUSBHostPort::disconnect: persistent enumeration failures
[116708.639602]: [IGFB][LOG  ][DISPLAY   ] ProAudioTest: proAudioWorkLoad notification timestamp = 116708639585028, state = Enable wake
[116708.831929]: 116708.831927 AppleUSB20XHCIPort@01400000: AppleUSBHostPort::disconnect: persistent enumeration failures
[116708.996781]: LQM-WiFi:TX(D2:21:F9:9F:14:D2) AC<SU MS NB NRS NA CM EX TF FFP MRET FLE> BE<29 0 0 0 0 0 0 0 0 0 0> (5000ms)
[116708.996792]: LQM-WiFi:TX(D2:21:F9:9F:14:D2) AC<SU MS NB NRS NA CM EX TF FFP MRET FLE> BK<0 0 0 0 0 0 0 0 0 0 0> (5000ms)
[... 60 lines cut ...]
[116724.641511]: [IGFB][LOG  ][DISPLAY   ] ProAudioTest: proAudioWorkLoad notification timestamp = 116724641494637, state = Enable wake
[116724.664883]: 116724.664880 AppleUSB20XHCIPort@01400000: AppleUSBHostPort::disconnect: persistent enumeration failures
[116725.315015]: 116725.315013 AppleUSB20XHCIPort@01400000: AppleUSBHostPort::disconnect: persistent enumeration failures
[116725.964712]: 116725.964683 AppleUSB20XHCIPort@01400000: AppleUSBHostPort::disconnect: persistent enumeration failures
[116726.042212]: [IGFB][LOG  ][DISPLAY   ] ProAudioTest: proAudioWorkLoad notification timestamp = 116726042189606, state = Disable Wake
[116726.615365]: 116726.615363 AppleUSB20XHCIPort@01400000: AppleUSBHostPort::disconnect: persistent enumeration failures
[116726.641629]: [IGFB][LOG  ][DISPLAY   ] ProAudioTest: proAudioWorkLoad notification timestamp = 116726641613131, state = Enable wake
[116726.772598]: tx_flush:1182: disk1 xid 24945915 tx stats: # 98720 finish 92212 enter 419 wait 9919 1791us close 48us flush 1616us
[116726.862427]: AppleKeyStore:10727:103: operation failed (sel: 75 ret: e00002ce, -1, 100000)
[116727.264855]: 116727.264849 AppleUSB20XHCIPort@01400000: AppleUSBHostPort::disconnect: persistent enumeration failures
[116727.914436]: 116727.914431 AppleUSB20XHCIPort@01400000: AppleUSBHostPort::disconnect: persistent enumeration failures
[116728.042668]: [IGFB][LOG  ][DISPLAY   ] ProAudioTest: proAudioWorkLoad notification timestamp = 116728042550616, state = Disable Wake
[116728.564108]: 116728.564103 AppleUSB20XHCIPort@01400000: AppleUSBHostPort::disconnect: persistent enumeration failures
[116728.642131]: [IGFB][LOG  ][DISPLAY   ] ProAudioTest: proAudioWorkLoad notification timestamp = 116728642037251, state = Enable wake
[116728.998800]: LQM-WiFi:TX(D2:21:F9:9F:14:D2) AC<SU MS NB NRS NA CM EX TF FFP MRET FLE> BE<20 0 0 0 0 0 0 0 0 0 0> (5000ms)
[116728.998827]: LQM-WiFi:TX(D2:21:F9:9F:14:D2) AC<SU MS NB NRS NA CM EX TF FFP MRET FLE> BK<0 0 0 0 0 0 0 0 0 0 0> (5000ms)
[116728.998839]: LQM-WiFi:TX(D2:21:F9:9F:14:D2) AC<SU MS NB NRS NA CM EX TF FFP MRET FLE> VO<0 0 0 0 0 0 0 0 0 0 0> (5000ms)
[116728.998871]: LQM-WiFi:L3 Control VO TX(D2:21:F9:9F:14:D2) Success=0 NoACK=0 Expired=0 OtherErr=0
[116728.998915]: LQM-WiFi:TX(FF:FF:FF:FF:FF:FF) AC<SU MS NB NRS NA CM EX TF FFP MRET FLE> BE<0 0 0 0 0 0 0 0 0 0 0> (5000ms)
[116729.214968]: 116729.214963 AppleUSB20XHCIPort@01400000: AppleUSBHostPort::disconnect: persistent enumeration failures
```

**Desktop (please complete the following information):**
 - OS: macOS 11.6.7
 - Chrysalis Version: [e.g. 0.4.0]: Not sure how to get this, but I have a brand-new Model 100 that I received this week.

richarddli, 2022-09-27:
On a whim, I connected an M2 Macbook Air running 12.5.1 to the exact docking station and I don't have a problem. (The original Mac is a 2020 13" Macbook Pro running Intel i5 processors).

richarddli, 2022-09-30:
I upgraded my Mac Intel to macOS 12.6 and still have the sleep problem.

richarddli, 2022-10-14:
I updated to 0.90.6.snapshot.55 and still see the issue.

tlyu, 2022-10-14:
Could you please describe the exact cabling setup for this situation? What type of cables, including types of connectors? How long are they? What hubs are being used? What kind of port is the Model 100 plugged into? USB-C? USB-A? Any adapters?

Does the Model 100 otherwise work as a keyboard?

If you select an animated LED mode on the keyboard, and try to put the Mac to sleep, do the LEDs go out? Do they stay lit? If so, does the animation freeze?

richarddli, 2023-01-19:
I switched computers so no longer see the issue.

closed 2024-02-26 by obra