keyboardio/Chrysalis #1097: Firmware update confirmation box tells you to click a button that's not there

aaronwe, 2022-09-27:
STEPS: Update firmware on your keyboard. The "Are you sure you want to update the firmware" dialog box appears.

[image]

ISSUE: The dialog box says "Continue holding it while you click the Update button." There is no Update button on this dialog box. There is a Continue button. This makes it unclear when the user is supposed to start holding the PROG key. (Will there be an Update button later, or do I need to start pressing PROG now?)

EXPECT: Consistency between instructions and button text. Either the Continue button becomes an Update button, or the dialog box text says "while you click the Continue button." (Or "BEGIN UPDATE" would make it exceptionally clear what happens next.)

obra, 2022-09-28:
Thanks! We've got this fixed for 0.11.5.

algernon, 2022-09-29:
Fixed in 0.11.5, closing.

closed 2022-09-29 by algernon