keyboardio/Chrysalis #1103: The Firmware Update page is getting a bit too high

algernon, 2022-09-27:
[image]

With full screen 1920x1080 window, the firmware update page is too high to show the steppers underneath without a scrollbar. This makes it a little bit more annoying to see how the update is progressing. We should do something about that.

An easy fix would be to move the stepper to the top, and push the rest of the screen down. Not ideal, but quick & easy.

obra, 2022-09-28:
We can also save a bunch of space by sticking the latest version number on
the same line as 'Latest Version' and moving the explanation of what a
locally built firmware needs into the target of the "Learn more" link.

tlyu, 2022-09-28:
Once the selections are made and the update initiated, I think it doesn't make sense to leave all of that instruction verbiage on the screen. Once the update has begin, maybe clear most of the text and show only the progress indicator?

algernon, 2022-09-28:
I like that idea too. We could perhaps roll it all up under "Current firmware version"... might even update that post-flash, as a small bonus?

tlyu, 2022-09-28:
On macOS 10.15.7 (19H2026), with an effective 1680x1050 screen size, the default window size makes most or all of the progress bar invisible without scrolling. I think this is a usability regression, especially if the scroll bars are hidden, which they often are on laptops.

algernon, 2022-09-28:
Fixed by #1104