keyboardio/Chrysalis #1107: Chrysalis loses connection to keyboard but doesn't notify / update the UI

aconbere, 2022-09-28:
**Describe the bug**
When running Chrysalis for longer periods of time in the background, when returning to the application the UI appears to be in a good state, however subsequent actions fail to render correctly (at least once the result was just a plain white page). Refreshing the page (Cmd-r) returns to the "select keyboard" page and everything works fine after that. Closing and quitting the app also restores functionality.

**To Reproduce**
Steps to reproduce the behavior:
1. Open Chrysalis and connect it to a keyboard
2. Wait a while (Order of hours)
3. Try doing something like viewing the layout card
4. OH NO!

**Expected behavior**
I think in an ideal case if Chrysalis is aware of the lost connection that it would either pop up a modal letting me know what's happened or just returned me to the "select keyboard" screen already.

**Desktop (please complete the following information):**
 - OS: MacOS 12.5.1
 - Chrysalis Version: 0.11.4 (0.11.4)

aconbere, 2022-09-29:
[image]
[image]

aconbere, 2022-09-29:
[chrysalis-debug.json.txt](https://github.com/keyboardio/Chrysalis/files/9676272/chrysalis-debug.json.txt)

algernon, 2022-09-29:
From the logs, it looks like we are, indeed, reporting the disconnect properly, so I suspect that the re-render might be responsible for the errors, and by way of the uncaught exception, for not finishing the handler properly.

algernon, 2022-09-29:
Ok, quick experiment shows that at least on Linux, we do fire an event when the keyboard disconnects. But it never does anything, we never get past these checks:

```
    if (!focus.focusDeviceDescriptor) return;
    if (flashing) return;

    if (
      focus.focusDeviceDescriptor.usb.vendorId != vid ||
      focus.focusDeviceDescriptor.usb.productId != pid
    ) {
      return;
    }
```

I assume the same thing happens on other platforms. Which means we're not handling disconnects properly. Anything else that runs after the `handleDeviceDisconnect` is thus running with false assumptions.

algernon, 2022-09-29:
for some odd reason, `focus.focusDeviceDescriptor` is null there. I don't yet know why.

algernon, 2022-09-29:
Hrm. `focus.close()` nulls that out, so now I have to figure out where that is called from, and we should be good.

algernon, 2022-09-29:
Oooh! We have an `on("close")` handler on the port in `api/focus`, which does call `focus.close()`.

PR #1115 (2022-09-29): api/focus: Do not close the port when receiving a close event
Using `this.close()` clears properties that we need elsewhere, in `handleDeviceDisconnect()` in App.js. With those properties cleared, the handler over there can't do its job properly. If it can't do its job properly, then we're not handling the disconnect either, which leaves other parts of Chrysalis in a bad state, where they assume the keyboard is connected while it is not.

Since we do have a way to handle disconnects, at a higher level, lets let it do that, and not close the device at a lower level.

Fixes #1107.
commit 0740be42 (2022-09-29): api/focus: Do not close the port when receiving a close event

Using `this.close()` clears properties that we need elsewhere, in
`handleDeviceDisconnect()` in App.js. With those properties cleared, the handler
over there can't do its job properly. If it can't do its job properly, then
we're not handling the disconnect either, which leaves other parts of Chrysalis
in a bad state, where they assume the keyboard is connected while it is not.

Since we do have a way to handle disconnects, at a higher level, lets let it do
that, and not close the device at a lower level.

Fixes #1107.
closed 2022-09-29 by obra