keyboardio/Chrysalis #1108: The Model100 firmware update news should more strongly suggest doing a factory reset up front

algernon, 2022-09-28:
(suggestion lifted from https://community.keyboard.io/t/model-100-firmware-update-to-fix-corruption-issues/5553/6)

tlyu, 2022-09-29:
Possibly we should go even further to suggest always doing the trick of the holding down PROG while plugging in the keyboard, while Chrysalis is running. This might help bypass any frustrations with communications being so unreliable that Chrysalis can't back up or reset the keyboard.

algernon, 2022-09-29:
I updated the forum post with the update instructions: it now shows doing the PROG trick, and explains the whole procedure with screenshots.

The breaking news text is fine, I believe. So closing this one.

closed 2022-09-29 by algernon