keyboardio/Chrysalis #1112: flashing while connected in bootloader mode doesn't erase EEPROM when requested

tlyu, 2022-09-29:
**Describe the bug**
One workaround for poor communications with Chrysalis and the factory Model 100 firmware is to force bootloader mode before connecting in Chrysalis. Unfortunately, if you then select the Factory Reset option when prompted to upgrade the firmware, the EEPROM does not get erased, despite the UI feedback implying that it will.

**To Reproduce**
Steps to reproduce the behavior:
1. Start Chrysalis without the keyboard connected.
2. Plug in a Model 100 keyboard while holding down PROG, to force bootloader mode.
3. Connect to the keyboard in Chrysalis.
4. Click Upgrade Firmware when prompted.
5. Select Factory Reset, and click the prompts to continue.
6. Observe that flashing completes, but only performs a "flashing" step.
7. Reconnect to the keyboard in Chrysalis.
8. Observe that any previous custom keymap layout settings remain intact.

**Expected behavior**
"Factory Reset" should behave as advertised, even when using the "connect to Chrysalis while in bootloader mode" workaround for poor connectivity with factory firmware.

**Screenshots**
N/A; there are lots of steps involved, and the description is hopefully clear enough to replicate the failure.

**Desktop (please complete the following information):**
 - OS: macOS 10.15.7 (19H2026)
 - Chrysalis Version: 0.11.5, but also 0.11.4 and maybe earlier as well.
 
**Additional context**
I came across this while trying to provide more robust steps for reflashing from Model 100 factory firmware in #1108.

algernon, 2022-09-29:
I had an attempt at this, but it's Difficult™, for a number of reasons. The major one being that `Model100` and `Model100Bootloader` are separate things, so if we're in bootloader mode, reconnecting to a normal mode keyboard does not currently work. This is one of the issues highlighted in #1015.

We also have a couple of other stability problems. It's hard to describe them, because it's complicated and messy, so I'd like to discuss it on Discord instead.

algernon, 2022-09-29:
As an interim solution, I'd remove the factory reset switch if the keyboard is in bootloader mode, to not set false expectations.

algernon, 2022-09-29:
...but doing that temporary removal isn't trivial either, for similar reasons. *sigh*

PR #1144 (2022-10-06): FirmwareUpdate: Rework how we do factory resets
The major part of this pull request is a rework of how we do factory resets. Originally we were erasing the EEPROM *before* going into bootloader mode, because that was easy to do. Unfortunately, it is also the wrong thing to do, because that prevents us from performing the same feat when starting from bootloader mode. We really want to do the erase after flashing new firmware, and before restoring the previous settings.

This pull request achieves that, and does a bit more under the hood, because those changes were necessary to be able to rearra [... cut]
commit 4957b474 (2022-10-06): FirmwareUpdate: Rework how we do factory resets

We used to do factory resets *before* flashing, because doing so was easy.
However, doing so is wrong, because that does not let us perform the feat when
started from bootloader mode. We want to be able to do that, too.

As such, we'll do the factory reset after flashing, right before restoring
EEPROM.

While doing the rearrangement, I also fixed an issue that prevented us from
removing `Model100Bootloader`: we did not correctly remember if we started from
bootloader mode. Now we do, so the extra descriptor is no longer necessary.

Fixes #1112.
closed 2022-10-06 by obra