keyboardio/Chrysalis #1114: Model100: LED Brightness not found

freeo, 2022-09-29:
**Describe the bug**
Judging by the Wiki and v0.7.0 release notes, there should be support for changing the LED brightness on the Model 100. I tested with Chrysalis v0.11.6-snapshot.137, v0.11.5 and also v0.7.0 (which doesn't find my firmware 0.90.2+44 keyboard). Because of that the Model 100 is locked at 100% LED brightness, which of course is crazy when using it at night.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to Preferences -> 'My Keyboard'
2. Look at "LEDs"
3. "LED Brightness" not found (unlike what the Wiki states)

**Expected Screenshot**
[image]

**Actual Screenshot**
[image]

**Desktop (please complete the following information):**
 - OS: Win 10.0.19044 
 - Chrysalis Version: v0.11.6-snapshot.137 (most current), v0.11.5 (current stable)
  
**Additional Information**
I'm not hiding "unsupported features" in my preferences.

tlyu, 2022-09-29:
The `FocusLEDCommand` plugin isn't active in the default Model 100 sketch. It probably should be.

tlyu, 2022-09-29:
One quirk to note is that the brightness setting doesn't currently seem to persist across reboots of the keyboard.

algernon, 2022-09-29:
Indeed, it does not. I think I'd prefer writing a new plugin that does persistence too, using the same command as `FocusLEDCommand`, because I'd like to avoid adding EEPROM storage to a plugin that didn't have such a dependency before.

closed 2022-10-21 by obra