keyboardio/Chrysalis #1119: Chrysalis will not save to my Model 01

sixears, 2022-09-29:
Chrysalis 0.11.5 never un-greys the "download" button to allow me to save changes to my Model01

**To Reproduce**
1. Go to Layout & Colormap Editor
2. Make a change to a key
3. Bounce like a crazed gibbon on the save/download button in the bottom-left corner.  It is a pleasing but futile light grey colour.

**Expected behavior**
Once I had changed a key, I would expect the download/save button to un-grey.

**Desktop (please complete the following information):**
 - OS: nixOS 22.05
 - Chrysalis Version: 0.11.5

sixears, 2022-09-29:
btw, Chrysalis prompted me to updated firmware; which I did successfully.
The firmware is now 0.90.2+44 .
It wouldn't let me save before I updated; reasoning that I needed to update fw to save, I went ahead.  I'm afraid I didn't note which firmware I was previously using.

obra, 2022-09-29:
Now that you've updated, is it working ok?

algernon, 2022-09-29:
Oooh. I think I know the problem. It would be the `layerNames` bounds check. It's probably failing on keyboards that do not have the `LayerNames` plugin enabled, such as the Model 01. I'll post a PR fixing it in a moment.

mattmc3, 2022-09-29:
I can confirm - I have the same issue on my Model 01.

closed 2022-09-29 by obra

sixears, 2022-09-30:
> Now that you've updated, is it working ok?

No, the update had no effect. I had made the update before I posted the issue; I was back-filling the info. about having tried the fw, to no avail.

Sorry, I realize I could've made that much clearer.

sixears, 2022-09-30:
I can confirm that the fix, as of 0.11.6-snapshot.141, is working for me.  Thank you.

PR #1120 (2022-09-29): Editor: Do not disable saving if LayerNames is not in use
In case the `LayerNames` plugin is not enabled firmware-side, getting the stored size of the layer names still returns a non-zero value. We need to check if there is *any* storage size, before comparing the serialized length, otherwise we mistakenly disable saving (and don't even show a warning, because the warning correctly checks for the presence of the plugin).

Fixes #1119.
commit 897f5de6 (2022-09-29): Editor: Do not disable saving if LayerNames is not in use

In case the `LayerNames` plugin is not enabled firmware-side, getting the stored
size of the layer names still returns a non-zero value. We need to check if
there is *any* storage size, before comparing the serialized length, otherwise
we mistakenly disable saving (and don't even show a warning, because the warning
correctly checks for the presence of the plugin).

Fixes #1119.