keyboardio/Chrysalis #1138: Incorrect input when quickly moving between layers

sQVe, 2022-10-03:
**Describe the bug**
I've been running the Model 100 as my daily driver for a couple of days and I've been frustrated by it sometimes inputting incorrectly. 

I have, for multiple years, been running a symbol layer when I hold down <kbd>Enter</kbd> without any issues. This is something the Model 100 doesn't like when inputting quickly while moving from layer to layer.

**To Reproduce**
Steps to reproduce the behavior:
1. Create a new layer.
2. Map <kbd>V</kbd> to `<` on new layer.
3. Map <kbd>B</kbd> to `>` on new layer.
4. Map <kbd>Enter</kbd> to `Enter` and layer shift when held to new layer.
5. Type `<div>` as fast possible while while using the new layer.

**Actual behavior**
It very often types:

```text
<div
b
```

https://user-images.githubusercontent.com/2284724/193691931-ffb87f9e-a726-4389-8806-24a9e6b3f31c.mp4

**Expected behavior**
It should type:

```text
<div>
```

**Desktop (please complete the following information):**
 - OS: Arch Linux (5.19.12-arch1-1)
 - Chrysalis Version: 0.11.5
 - Firmware version: 0.90.3+46

**Additional context**
Just to be clear, this is the sequence: `Enter (hold)`, `Press b`, `Enter (release)`, `Press d`, `Press i`, `Press v`, `Enter (hold)`, `Press v`.

anderium, 2022-10-03:
This sounds like you are typing faster than the default settings of Qukeys are expecting you to.  I think you may want to play around with some of the parameters mentioned in the [Qukeys documentation](https://kaleidoscope.readthedocs.io/en/latest/plugins/Kaleidoscope-Qukeys.html).  Specifically I think you may want to switch layers within 75ms of typing the final v of div, so you want to lower the [`MinimumPriorInterval`](https://kaleidoscope.readthedocs.io/en/latest/plugins/Kaleidoscope-Qukeys.html#setminimumpriorinterval-min-interval).  I think you cannot yet set these values from within Chrysalis, so you'll have to compile your own sketch.  (You **will** be able to use this sketch in Chrysalis after compiling it once.)

PS: Are you sure you're typing what you say?  From what you describe I would think that should show `<divb\n`

sQVe, 2022-10-03:
@anderium Yeah, the output should be correct. It's making an newline before the `b`. It's like it can't handle the enter hold since it's going too quickly.

I'll have to test tweaking those settings tomorrow. Thanks for the suggestion.

sQVe, 2022-10-03:
@anderium Could you possibly point me in the right direction on how I would just alter this value on top of the default firmware? 

I'm a bit unsure about the setup coming from QMK land. All these plugins are confusing.

anderium, 2022-10-03:
Unfortunately I can't be of much help with that, I'm still eagerly awaiting my own keyboard.  But to give you some pointers, you can probably follow the steps [here](https://kaleidoscope.readthedocs.io/en/latest/quick_start.html) ([installing Arduino](https://kaleidoscope.readthedocs.io/en/latest/setup_toolchain.html) and [building firmware](https://kaleidoscope.readthedocs.io/en/latest/build_default_firmware.html)).  I think you may encounter an outdated dependency (Arduino BSP?) that has a required fix on master, but not release.  Other people have also been having problems with their flashing process, so you may find the same problem and an answer on https://community.keyboard.io/ or the Discord.  And in the end, if flashing is successful you will still see error code 251, which is a bug in dfu-util.

There's also this, which I found while looking around:
> Flashing via Arduino isn't working at the moment, because Arduino doesn't know how to put the keyboard into programmable mode yet.  As a workaround, you can unplug your keyboard, hold the PROG key while plugging it in, and then flash

sQVe, 2022-10-04:
@anderium I followed your suggestions and it worked awesomely. I really appreciate you taking your time trying to help. 🙏🏼 

Setting `Qukeys.setMinimumPriorInterval(0);` gives me a consistent good result. I just compile it to a binary and then flash via Chrysails. Easy peasy.

closed 2022-10-04 by sQVe

Rom1deTroyes, 2022-10-04:
Related to #324 (for the record) ?