keyboardio/Chrysalis #1141: Artifacts on corners of keycaps in "Layout & Colormap Editor"

Pattrigue, 2022-10-04:
**Describe the bug**
The strokes around the keycaps in the Layout & Colormap Editor has a few noticeable artifacts around the edges.

**To Reproduce**
Simply open Chrysalis with a Model 100 connected and click connect. Should be visible at once in the editor view.

**Expected behavior**
There should not be any artifacts - the corners should be rounded.

**Screenshots**
[image]
[image]

**Desktop (please complete the following information):**
 - OS: Windows 10 64-bit
 - Chrysalis Version: 0.11.6

**Additional context**
I use a 4k monitor. Perhaps this issue is not visible on 1080p?
Seems more prominent on the right side for some reason. It's barely visible on the left side, as shown in the second screenshot.

tlyu, 2022-10-04:
It looks like some of the SVG paths aren't actually closed, but I haven't diagnosed in detail yet.

tlyu, 2022-10-06:
After some further investigation, it seems like some of the paths that show artifacts are indeed closed, but there might be a small distance between the last and first points of the path. This is possibly causing a very short line segment between those points, with very sharp joint angles and excessively long, spiky miters.

As a workaround, maybe set `stroke-linejoin="round"` on the paths? Ultimately, we should correct the paths to either have an exact overlap, or to use the closing path segment to replace an existing straight line segment in the path.

tlyu, 2022-10-26:
I've submitted a pull request that should fix most of these (at least 3 of the ones noted here, as far as i can see on my Retina MBP). The fourth thumb key on the left hand might still have a 1-pixel artifact or so, but maybe it's less visible now? It was being resistant to my simple efforts to correct it, so a more aggressive approach may be needed.