keyboardio/Chrysalis #1161: Fix ui theme preference 'system' for Linux

EvyBongers, 2022-10-09:
**Is your feature request related to a problem? Please describe.**
Dark mode preference `system` doesn't follow my system preference (Linux).

**Describe the solution you'd like**
On Linux, follow [`org.freedesktop.appearance.color-scheme`](https://github.com/flatpak/xdg-desktop-portal/blob/d7a304a00697d7d608821253cd013f3b97ac0fb6/data/org.freedesktop.impl.portal.Settings.xml#L33-L45) when dark mode preference is set to `system`.

**Describe alternatives you've considered**
Wait for Chromium/Electron to [implement support](https://bugs.chromium.org/p/chromium/issues/detail?id=1340411). :)

**Additional context**
Mostly created this ticket to start a discussion. This issue is true for any electron based app as well as chromium/chrome.

EvyBongers, 2022-10-09:
My guess would be that Gnome (also) updates the gtk setting for it, which Chrysalis does follow.

obra, 2024-02-26:
Chrysalis now runs in-browser using WebSerial and WebUSB, so I'm closing out this issue as obsolete.