keyboardio/Chrysalis #1173: Add layer names to backup and restore

EvyBongers, 2022-10-12:
**Is your feature request related to a problem? Please describe.**
Layer names aren't saved when backing up keyboard layout.

**Describe the solution you'd like**
When backing up keyboard layout, the layer names should also be saved.

**Describe alternatives you've considered**
-

**Additional context**
-

obra, 2024-02-26:
They are now added to backups, but are not added to restores. Updating this issue accordingly

parke, 2024-08-23:
I've noticed that as of Chrysalis 2024.0815.1901, at least some secondary key functions are displayed differently when I select `Backup and Restore layouts` in the hamburger menu.

For example, when I am in the Editor, the secondary key function will display `MOV/` (i.e. temporarily shift to the layer named `MOV`).

Whereas when I switch to `Backup and Restore layouts`, the same key will unexpectedly display as its secondary function as `Layer #3/`.

Would you like me to report this discrepancy as a new issue?  Or perhaps the current (unexpected) behavior is correct?

Also, when I click the `Backup to file...` button, there is an 11 second delay before I am prompted to save the file.  Is this normal?  (The delay might be a result of some quirk in my admittedly highly non-standard Linux setup.  I have not tried to reproduce the delay on one of my Chromebooks.)