keyboardio/Chrysalis #1188: Chrysalis ^v0.9.4 not rendering(?) windows 11

aguthrie19, 2022-10-17:
**Describe the bug**
Chrysalis v0.9.4 is the last release that properly starts when run, other releases like v0.9.4 and v0.11.8 briefly flash a window and then disappear.

**To Reproduce**
Steps to reproduce the behavior:
1. Download [Chrysalis-0.9.4.exe](https://github.com/keyboardio/Chrysalis/releases/download/v0.9.4/Chrysalis-0.9.4.exe) and [Chrysalis-0.9.5.exe](https://github.com/keyboardio/Chrysalis/releases/download/v0.9.5/Chrysalis-0.9.5.exe)
2. Click on Chrysalis-0.9.4.exe, allow installation, open the app, window appears, can use as expected. Uninstall the app from Windows>Settings>Apps>Installed apps.
3. Click on Chrysalis-0.9.5.exe, allow installation, open the app, a window briefly flashes before disappearing. Noticed in task manager that the chrysalis process moves from the "Apps" to "Background Processes" and remains there.

**Expected behavior**
The chrysalis window to stay open.

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Windows 11 Home 22H2 build 22621.674
 - Chrysalis Version: ^v0.9.4

**Additional context**
- Perhaps culprit is in:
  - [src/main/index.js](https://github.com/keyboardio/Chrysalis/compare/v0.9.4...v0.9.5?diff=split#diff-f4ca67281c6be819c2166841d7631ffd5316d4c3a3b40e01a512c4113a537b2c)
  - [src/renderer/components/BoardMenu.js](https://github.com/keyboardio/Chrysalis/compare/v0.9.4...v0.9.5?diff=split#diff-eacf8a638c1eb413b070820d746f4b3e218b7f34407a0d4e101d62195003da36)
  - [src/renderer/screens/KeyboardSelect.js](https://github.com/keyboardio/Chrysalis/compare/v0.9.4...v0.9.5?diff=split#diff-d1da75dd7b7b9f8689fd0f8b8b9b95cee6132f02225dae955280c8eba99d8638)
- I've removed Microsoft Windows Edge from my machine (AveYo's [Edge_Removal.bat](https://github.com/AveYo/fox/blob/main/Edge_Removal.bat)) so perhaps this _plus_ chrysalis changes create rendering problems. Not really sure its a rendering problem though?

aguthrie19, 2022-10-18:
Nothing in the log
[image]

aguthrie19, 2022-10-25:
With a clean windows installation, not using AveYo's [Edge_Removal.bat](https://github.com/AveYo/fox/blob/main/Edge_Removal.bat), then chrysalis v0.9.5 and v0.11.8 work.
@algernon Does chrysalis depend on processes Win32WebViewHost or MSEdgeHTM?

lemuus, 2024-01-12:
I'm having exactly the same issue. I can run correctly up to 0.9.4. But after that is just a blank screen that flash and close, and I can see the process lingering in task manager.
No logs from chrysalis.

This is on a win11 machine, and I haven't removed any windows component.

@algernon any suggestion? I also compiled from branch (v0.13.3 and current head) and I get she same behavior.

Thanks in advance :)

obra, 2024-01-18:
@lemuus - Can you try with https://chrysalis.keyboard.io? All future Chrysalis development is going toward the new browser-based tooll

lemuus, 2024-01-19:
@obra thanks!! that worked perfectly :)

obra, 2024-02-26:
Chrysalis now runs in-browser using WebSerial and WebUSB, so I'm closing out this issue as obsolete.

closed 2024-02-26 by obra