keyboardio/Chrysalis #1189: Communication timeout when connection to Model 1 on macOS

d6o, 2022-10-17:
**Describe the bug**
I downloaded Chrysalis using Homebrew, and when I try to connect to my model 01, it always gives me a `communication timeout` error. 

**To Reproduce**
Steps to reproduce the behavior:
1. Run `brew update && brew upgrade`
2. Run `brew install --cask chrysalis`
3. Open Chrysalis
4. Click on connect

**Expected behavior**
Chrysalis should connect to my Model 01

**Screenshots**
[image]
[image]

**Desktop (please complete the following information):**
 - OS: macOS Monterey 12.6 (M1 pro)
 - Chrysalis Version: 0.11.8

Logs:
```
{
  "timestamp": "2022-10-17T15:57:13.769Z",
  "uuid": "f72d8f51-2384-409e-ae09-ea8f2f64ed12",
  "chrysalis": {"version": "0.11.8"},
  "os": {"platform": "darwin", "distro": "macOS", "release": "12.6", "codename": "macOS Monterey", "kernel": "21.6.0", "arch": "arm64", "hostname": "Diegos-MacBook-Pro.local", "fqdn": "Diegos-MacBook-Pro.local", "codepage": "UTF-8", "logofile": "darwin", "serial": "778CC57A-CF0B-3D35-8EE8-5035142D0177", "build": "21G115", "servicepack": "", "uefi": true},
  "usb": [],
  "logs": [
    {"args": ["/keyboard-select"], "level": "debug", "message": "navigating to location", "timestamp": "2022-10-17T15:56:28.246Z"},
    {"function": "find", "label": "focus", "level": "debug", "message": "serial port list obtained", "portList": [{"path": "/dev/tty.Bluetooth-Incoming-Port"}, {"locationId": "00100000", "manufacturer": "Keyboardio", "path": "/dev/tty.usbmodemCkbio011", "productId": "2301", "serialNumber": "Ckbio01", "vendorId": "1209"}], "timestamp": "2022-10-17T15:56:28.270Z"},
    {"devices": [{"focusDeviceDescriptor": {"bootloader": false, "info": {"displayName": "Keyboardio Model 01", "product": "Model01", "urls": [{"name": "Homepage", "url": "https://shop.keyboard.io/"}, {"name": "Forum", "url": "https://community.keyboard.io/"}, {"name": "Chat", "url": "https://keyboard.io/discord-invite"}], "vendor": "Keyboardio"}, "keyboard": {"columns": 16, "rows": 4}, "usb": {"bootloader": {"productId": 8960, "protocol": "avr109", "vendorId": 4617}, "productId": 8961, "vendorId": 4617}}, "locationId": "00100000", "manufacturer": "Keyboardio", "path": "/dev/tty.usbmodemCkbio011", "productId": "2301", "serialNumber": "Ckbio01", "vendorId": "1209"}], "function": "find", "label": "focus", "level": "debug", "message": "supported devices found", "timestamp": "2022-10-17T15:56:28.270Z"},
    {"level": "info", "message": "Connecting to port", "path": "/dev/tty.usbmodemCkbio011", "timestamp": "2022-10-17T15:56:30.170Z"},
    {"level": "info", "message": "Probing for focus support...", "timestamp": "2022-10-17T15:56:30.189Z"},
    {"label": "focus", "level": "verbose", "message": "request", "request": {"args": [], "command": "help", "id": 0}, "timestamp": "2022-10-17T15:56:30.191Z"},
    {"function": "find", "label": "focus", "level": "debug", "message": "serial port list obtained", "portList": [{"path": "/dev/tty.Bluetooth-Incoming-Port"}, {"locationId": "00100000", "manufacturer": "Keyboardio", "path": "/dev/tty.usbmodemCkbio011", "productId": "2301", "serialNumber": "Ckbio01", "vendorId": "1209"}], "timestamp": "2022-10-17T15:56:33.284Z"},
    {"devices": [{"focusDeviceDescriptor": {"bootloader": false, "info": {"displayName": "Keyboardio Model 01", "product": "Model01", "urls": [{"name": "Homepage", "url": "https://shop.keyboard.io/"}, {"name": "Forum", "url": "https://community.keyboard.io/"}, {"name": "Chat", "url": "https://keyboard.io/discord-invite"}], "vendor": "Keyboardio"}, "keyboard": {"columns": 16, "rows": 4}, "usb": {"bootloader": {"productId": 8960, "protocol": "avr109", "vendorId": 4617}, "productId": 8961, "vendorId": 4617}}, "locationId": "00100000", "manufacturer": "Keyboardio", "path": "/dev/tty.usbmodemCkbio011", "productId": "2301", "serialNumber": "Ckbio01", "vendorId": "1209"}], "function": "find", "label": "focus", "level": "debug", "message": "supported devices found", "timestamp": "2022-10-17T15:56:33.284Z"},
    {"error": "Communication timeout", "label": "focus", "level": "error", "message": "request timed out", "request": {"args": [], "command": "help", "id": 0}, "timestamp": "2022-10-17T15:57:00.447Z"},
    {"device": {"accessible": true, "focusDeviceDescriptor": {"bootloader": false, "components": {}, "info": {"displayName": "Keyboardio Model 01", "product": "Model01", "urls": [{"name": "Homepage", "url": "https://shop.keyboard.io/"}, {"name": "Forum", "url": "https://community.keyboard.io/"}, {"name": "Chat", "url": "https://keyboard.io/discord-invite"}], "vendor": "Keyboardio"}, "keyboard": {"columns": 16, "rows": 4}, "usb": {"bootloader": {"productId": 8960, "protocol": "avr109", "vendorId": 4617}, "productId": 8961, "vendorId": 4617}}, "locationId": "00100000", "manufacturer": "Keyboardio", "path": "/dev/tty.usbmodemCkbio011", "productId": "2301", "serialNumber": "Ckbio01", "vendorId": "1209"}, "error": "Communication timeout", "level": "error", "message": "error while trying to connect", "timestamp": "2022-10-17T15:57:00.448Z"},
    {"label": "toast", "level": "debug", "message": "Toast message received", "msg": {"message": "Communication timeout", "variant": "error"}, "timestamp": "2022-10-17T15:57:00.449Z"},
    {"function": "find", "label": "focus", "level": "debug", "message": "serial port list obtained", "portList": [{"path": "/dev/tty.Bluetooth-Incoming-Port"}, {"locationId": "00100000", "manufacturer": "Keyboardio", "path": "/dev/tty.usbmodemCkbio011", "productId": "2301", "serialNumber": "Ckbio01", "vendorId": "1209"}], "timestamp": "2022-10-17T15:57:04.919Z"},
    {"devices": [{"focusDeviceDescriptor": {"bootloader": false, "info": {"displayName": "Keyboardio Model 01", "product": "Model01", "urls": [{"name": "Homepage", "url": "https://shop.keyboard.io/"}, {"name": "Forum", "url": "https://community.keyboard.io/"}, {"name": "Chat", "url": "https://keyboard.io/discord-invite"}], "vendor": "Keyboardio"}, "keyboard": {"columns": 16, "rows": 4}, "usb": {"bootloader": {"productId": 8960, "protocol": "avr109", "vendorId": 4617}, "productId": 8961, "vendorId": 4617}}, "locationId": "00100000", "manufacturer": "Keyboardio", "path": "/dev/tty.usbmodemCkbio011", "productId": "2301", "serialNumber": "Ckbio01", "vendorId": "1209"}], "function": "find", "label": "focus", "level": "debug", "message": "supported devices found", "timestamp": "2022-10-17T15:57:04.919Z"}
  ]
}```

RKJuve, 2022-11-02:
I'm also getting timeouts connecting to a Model 01 on macOS.  Using Chrysalis `0.12.0`

Hacklor, 2022-12-13:
I'm also getting the timeout for model 01 using Chrysalis 0.12.0 on Ubuntu 20.04 LTS.

It seems to detect the model on `/dev/ttyACM0` and I can use the keyboard. Chrysalis doesn't want to connect.

```
{
  "timestamp": "2022-12-13T14:29:49.001Z",
  "uuid": "0d708b02-20f4-4935-8a30-c38982724b0e",
  "chrysalis": {"version": "0.12.0"},
  "os": {"platform": "linux", "distro": "Ubuntu", "release": "20.04.5 LTS", "codename": "Focal Fossa", "kernel": "5.8.0-63-generic", "arch": "x64", "hostname": "nicole-zenbook", "fqdn": "nicole-zenbook", "codepage": "UTF-8", "logofile": "ubuntu", "serial": "b7f225c0cd674251a42b637bed7e65f2", "build": "", "servicepack": "", "uefi": true},
  "usb": [
    {"bus": 2, "deviceId": 4, "id": "07ab:fc73", "name": "", "type": "Storage", "removable": null, "vendor": "Freecom Technologies", "manufacturer": "", "maxPower": "0mA", "serialNumber": null},
    {"bus": 2, "deviceId": 3, "id": "0b95:1790", "name": "AX88179 Gigabit Ethernet", "type": "Vendor Specific Class", "removable": null, "vendor": "ASIX Electronics Corp.", "manufacturer": "", "maxPower": "184mA", "serialNumber": null},
    {"bus": 2, "deviceId": 2, "id": "2109:0817", "name": "", "type": "Hub", "removable": null, "vendor": "VIA Labs, Inc.", "manufacturer": "", "maxPower": "0mA", "serialNumber": null},
    {"bus": 2, "deviceId": 1, "id": "1d6b:0003", "name": "3.0 root hub", "type": "Hub", "removable": null, "vendor": "Linux Foundation", "manufacturer": "", "maxPower": "0mA", "serialNumber": null},
    {"bus": 1, "deviceId": 5, "id": "13d3:56cb", "name": "", "type": "Video", "removable": null, "vendor": "IMC Networks", "manufacturer": "", "maxPower": "500mA", "serialNumber": null},
    {"bus": 1, "deviceId": 4, "id": "248a:8367", "name": "", "type": "Human Interface Device", "removable": null, "vendor": "Maxxter", "manufacturer": "", "maxPower": "50mA", "serialNumber": null},
    {"bus": 1, "deviceId": 6, "id": "2109:8817", "name": "", "type": "", "removable": null, "vendor": "VIA Labs, Inc.", "manufacturer": "", "maxPower": "100mA", "serialNumber": null},
    {"bus": 1, "deviceId": 3, "id": "2109:2817", "name": "", "type": "Hub", "removable": null, "vendor": "VIA Labs, Inc.", "manufacturer": "", "maxPower": "0mA", "serialNumber": null},
    {"bus": 1, "deviceId": 7, "id": "8087:0026", "name": "", "type": "Wireless", "removable": null, "vendor": "Intel Corp.", "manufacturer": "", "maxPower": "100mA", "serialNumber": null},
    {"bus": 1, "deviceId": 8, "id": "1209:2301", "name": "Keyboardio Model 01", "type": "Keyboard", "removable": null, "vendor": "Generic", "manufacturer": "Keyboardio", "maxPower": "500mA", "serialNumber": null},
    {"bus": 1, "deviceId": 1, "id": "1d6b:0002", "name": "2.0 root hub", "type": "Hub", "removable": null, "vendor": "Linux Foundation", "manufacturer": "", "maxPower": "0mA", "serialNumber": null}
  ],
  "logs": [
    {"args": ["/keyboard-select"], "level": "debug", "message": "navigating to location", "timestamp": "2022-12-13T14:28:46.599Z"},
    {
      "function": "find",
      "label": "focus",
      "level": "debug",
      "message": "serial port list obtained",
      "portList": [{"manufacturer": "Keyboardio", "path": "/dev/ttyACM0", "pnpId": "usb-Keyboardio_Model_01_Ckbio01-if00", "productId": "2301", "serialNumber": "Ckbio01", "vendorId": "1209"}, {"path": "/dev/ttyS0"}, {"path": "/dev/ttyS1"}, {"path": "/dev/ttyS10"}, {"path": "/dev/ttyS11"}, {"path": "/dev/ttyS12"}, {"path": "/dev/ttyS13"}, {"path": "/dev/ttyS14"}, {"path": "/dev/ttyS15"}, {"path": "/dev/ttyS16"}, {"path": "/dev/ttyS17"}, {"path": "/dev/ttyS18"}, {"path": "/dev/ttyS19"}, {"path": "/dev/ttyS2"}, {"path": "/dev/ttyS20"}, {"path": "/dev/ttyS21"}, {"path": "/dev/ttyS22"}, {"path": "/dev/ttyS23"}, {"path": "/dev/ttyS24"}, {"path": "/dev/ttyS25"}, {"path": "/dev/ttyS26"}, {"path": "/dev/ttyS27"}, {"path": "/dev/ttyS28"}, {"path": "/dev/ttyS29"}, {"path": "/dev/ttyS3"}, {"path": "/dev/ttyS30"}, {"path": "/dev/ttyS31"}, {"path": "/dev/ttyS4"}, {"path": "/dev/ttyS5"}, {"path": "/dev/ttyS6"}, {"path": "/dev/ttyS7"}, {"path": "/dev/ttyS8"}, {"path": "/dev/ttyS9"}],
      "timestamp": "2022-12-13T14:28:46.721Z"
    },
    {"devices": [{"focusDeviceDescriptor": {"bootloader": false, "info": {"displayName": "Keyboardio Model 01", "product": "Model01", "urls": [{"name": "Homepage", "url": "https://shop.keyboard.io/"}, {"name": "Forum", "url": "https://community.keyboard.io/"}, {"name": "Chat", "url": "https://keyboard.io/discord-invite"}], "vendor": "Keyboardio"}, "keyboard": {"columns": 16, "rows": 4}, "usb": {"bootloader": {"productId": 8960, "protocol": "avr109", "vendorId": 4617}, "productId": 8961, "vendorId": 4617}}, "manufacturer": "Keyboardio", "path": "/dev/ttyACM0", "pnpId": "usb-Keyboardio_Model_01_Ckbio01-if00", "productId": "2301", "serialNumber": "Ckbio01", "vendorId": "1209"}], "function": "find", "label": "focus", "level": "debug", "message": "supported devices found", "timestamp": "2022-12-13T14:28:46.722Z"},
    {"level": "info", "message": "Connecting to port", "path": "/dev/ttyACM0", "timestamp": "2022-12-13T14:28:51.513Z"},
    {"level": "info", "message": "Probing for focus support...", "timestamp": "2022-12-13T14:28:51.571Z"},
    {"label": "focus", "level": "verbose", "message": "request", "request": {"args": [], "command": "help", "id": 0}, "timestamp": "2022-12-13T14:28:51.574Z"},
    {
      "function": "find",
      "label": "focus",
      "level": "debug",
[... cut ...] dor": "Keyboardio"}, "keyboard": {"columns": 16, "rows": 4}, "usb": {"bootloader": {"productId": 8960, "protocol": "avr109", "vendorId": 4617}, "productId": 8961, "vendorId": 4617}}, "manufacturer": "Keyboardio", "path": "/dev/ttyACM0", "pnpId": "usb-Keyboardio_Model_01_Ckbio01-if00", "productId": "2301", "serialNumber": "Ckbio01", "vendorId": "1209"}], "function": "find", "label": "focus", "level": "debug", "message": "supported devices found", "timestamp": "2022-12-13T14:29:26.724Z"}
  ]
}
```

Hacklor, 2022-12-20:
I'm also getting the timeout for model 01 using Chrysalis 0.12.0 on Windows 10. So it doesn't seem to be an issue related to the OS. Instead it's related to the model and Chrysalis version?

I would love to see this fixed, because right now I cannot use Chrysalis at all, and I would really like to change some key settings. Any idea on a timeline?

ldumancas, 2023-01-29:
I am also experiencing this connection issue on the latest version of the software (0.12). I rolled back to Chrysalis version 0.8.6 and was able to communicate with keyboard that way.

ldumancas, 2023-01-30:
after rolling the version of Chrysalis back to 0.8.6 I was able to flash the firmware of the Model 01. I then reinstalled version 0.12.0 and was able to connect to the keyboard and flash to the latest firmware. I no longer appear to be experiencing connectivity issues between my Model 01 and Chrysalis.

greenscar, 2023-04-22:
FWIW: I'm on a Model01. I tried Chrysalis 0.12.0 & 0.10.4; same error as above.
I bumped down to 0.8.6 and was able to flash my firmware. NOTE: if you don't hold the `prog` key, you'll error out.
Next, I opened 0.12.0 & connected successfully.
So it looks like either 0.9 or 0.10 included something to block communication with factory firmware.

obra, 2024-02-26:
Chrysalis now runs in-browser using WebSerial and WebUSB and this functionality has been rewritten, so I'm closing out this issue as obsolete. Please don't hesitate to open a new issue if https://chrysalis.keyboard.io exhibits the same behavior

closed 2024-02-26 by obra