keyboardio/Chrysalis #1193: NumLock point does't work anymore

necsoiu, 2022-10-22:
**Describe the bug**
NumLock point does't work anymore

**To Reproduce**
Press point (.) on NumLock 

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Windows 11
 - Chrysalis Version: 0.11.8

[chrysalis-debug.json.txt](https://github.com/keyboardio/Chrysalis/files/9843905/chrysalis-debug.json.txt)

obra, 2022-10-24:
If you map that physical key to another key (like the regular period key),
does it work?

Do you know what caused it to stop working? (What changed that caused it to
stop working)?

Best,
jesse

obra, 2023-01-06:
if you swap the two keyboards, what happens?

We really need to solve this issue. Strange thing is that on other Atreus with the same firmware version and same configuration it works. If needed I can provide debug files for both keyboards. In this moment both have version 0.12 of Firmware and are on the same SO (Windows 11).
Curios thing is that on the computer where it does't work I have other problem too which is reported here [https://github.com//issues/733]. I think they may be related and should be investigated together.
Another aspect is that one of the keyboards is older and another newer. Has something changed in the keyboard hardware that could lead to these two problems?

necsoiu, 2023-01-10:
On the HP computer both keyboards seems to work fine. So I guess it may be something that is not compatible with ASUS motherboards. Do you think it can be solved?

necsoiu, 2023-02-12:
OK. So today I solved the problem, but I haven't fully identified what caused the problem to be solved so I will detail the steps I took.
1. I made a motherboard BIOS update. (then I tested but the problem still exists).
2. I pluged a comercial keyboard to test if works (with this keyboard everything worked well, but with Atreus still don't)
3. I locked the Atreus to first level to test if on this level the numpad dot works (surprise, it worked).
4. Reverted back Atreus to layer 0 and surprise, surprise, it worked here too. 

So in the end I think it could be the 
1. The BIOS
2. Or the comercial keyboard somehow unlocked something
3. Or changing Atreus on layer 1 somehow unlocked something. I mention that on layer 0 the numpead keys ar accessed by pressing a layer shift key (to le leyer 1).

Anyway I'm super happy that this problem was solved.

closed 2024-11-11 by obra