keyboardio/Chrysalis #1203: My Keyboard: LED Profile: Instructions Unclear

Zian, 2022-10-24:
**Describe the bug**
The numerical picker that starts at 255 does not appear to correspond to any particular LED profile. It is not clear how to select the desired LED profile.

**To Reproduce**
Steps to reproduce the behavior:
1. Plug in a Keyboardio 100.
2. Start Chrysalis.
3. Click CONNECT
4. Click on the hamburger menu.
5. Click on Preferences.
6. Click on MY KEYBOARD.
7. Scroll down to the section labelled "LEDs".
8. Look at the item named "Default led mode".

**Expected behavior**
I expected to see a way to select the LED profile currently in use or a way to select it from the list used when I press the "led" key on the keyboard.

Instead, the program showed a numerical selector set to "255".

**Screenshots**
[image]

[chrysalis-debug.json.txt](https://github.com/keyboardio/Chrysalis/files/9847982/chrysalis-debug.json.txt)

**Desktop (please complete the following information):**
 - OS: Windows 10 Version 21H2
 - Chrysalis Version: 0.11.9
**Additional context**
Add any other context about the problem here.

danielcranford, 2022-10-26:
I figured how 'Default LED mode' works, but it really needs better documentation. Each LED effect is given an ordinal with 0 being 'off' and 14 being the colors defined in the keymap editor (at least on my model 100). The number of times you have to press the 'LED Next' key to get to your desired default effect is the number to put into the field.

Zian, 2022-11-08:
Would the following implementation be feasible? I don't know how much information Chrysalis can get from the keyboard.

UI: Live Preview like the Style picker in Microsoft Office Word (LED preview on-hover). Also show the ID # just in case someone's documented/memorized their LED #.
On-click: apply and save setting

Zian, 2022-11-08:
> I figured how 'Default LED mode' works, but it really needs better documentation. Each LED effect is given an ordinal with 0 being 'off' and 14 being the colors defined in the keymap editor (at [...]

After reading your explanation, I still had trouble. The Keyboardio 100 turns all the lights off if you push the led key after reaching the mode that shows a blue light behind all the keys. However, this mode-after-the-blue-lights is not actually mode 0. If you start typing, keys will light up.

Instead, you should continue to press the led key until the keys light up again. Now, you are on mode 1. The mode that preceded it with all the lights off is the true "off" or "0" mode.