keyboardio/Chrysalis #1204: Space (right thumb button) stopped working and became the same as the one next to it midsentence.

tonycassara, 2022-10-24:
**Describe the bug**
I was typing a sentence and the space bar stopped working. It suddenly was tabbing instead of space, which is the key directly to the right of it. Space is currently 2nd from right thumb button, Tab is far right thumb button. I had to disconnect and reconnect to fix it, no flashing necessary.

**To Reproduce**
Steps to reproduce the behavior:
Happened randomly mid sentence typing a comment on YouTube in Microsoft Edge Version 106.0.1370.52 (Official build) (64-bit)
Unable to faithfully reproduce (yet).

**Expected behavior**
Space bar would make a space, not tab out.

**Screenshots**
If applicable, add screenshots to help explain your problem. If possible, please also include the debug bundle you can create on the "Report a problem" screen. If, for some reason, you are unable to access that screen, please press `Control + Shift + I` (Windows or Linux) or `Command + Option + I` (macOS) to bring up the developer console. Right-clicking inside it will allow you to save its contents to a file - please attach that to the bug report.

**Desktop (please complete the following information):**
 - OS: Windows 11 22621.674
 - Chrysalis Version: 0.11.9

**Additional context**
From Chrysalis shortly after the bug, it happened to be running when it happened and during the keyboard unplug/replug in so hopefully this is useful: https://www.dropbox.com/s/1okrx2gp6yc24vl/chrysalis-debug.json.txt?dl=0

FrankyGee, 2022-10-25:
This is a long shot, but are you connected remotely?
I'm having something quite similar with braces `{}` appearing as square brackets `[]` on 9/10 attempts, _but only intermittently when connected to a remote system with a high lag_. So I'll mash `{` a bunch of times and get `[[[[[[[{`

tonycassara, 2022-10-26:
No unfortunately this is on my local computer, not a remote session. That is very interesting though!

tonycassara, 2022-11-03:
For anyone else who has this issue I accidentally enabled locking to layer 1 which had a random key code 655 applied where I use space.

closed 2022-11-03 by tonycassara