keyboardio/Chrysalis #1218: Error while updating firmware on Model 100 on Windows.

e-max, 2022-10-29:
**Describe the bug**
Error while updating firmware on Model 100 on Windows.

**To Reproduce**
Steps to reproduce the behavior:
1. Start Chrysalis
2. Select "Keyboardio Model 100 (Unknown)" and press "Connect"
3.  It displays "Chrysalis recognizes your keyboard but needs to update firmware".   Press "Update firmware"
4.  Choose "Latest version (0.91.0)".   Do not choose  "Erase everything".   Do not press the "Prog" button.  (Well the error happens with or without this button)
5. See error "Error flashing the firmware" 

**Expected behavior**
The firmware was successfully updated.

**Screenshots**

[image]

The error bundle is attached. 

**Desktop (please complete the following information):**
 - OS:  Windows 8.1
 - Chrysalis Version:  0.12.0

**Additional context**

It's not the first time I am trying to update the firmware.  The first time I did it on MacOS and that time it was a success.  I ran Chrysalis after and it recognized my keyboard, allowed me to change settings, and so on.
   Unfortunately, I don't have access to Macbook anymore.  So I tried the keyboard with Windows.   The keyboard was recognized by Windows, I can type and so on, but when I run the latest Chrysalis it says that I must update a firmware.  And when I try to do it, it fails.

e-max, 2022-10-29:
[chrysalis-debug.json.txt](https://github.com/keyboardio/Chrysalis/files/9894103/chrysalis-debug.json.txt)

tlyu, 2022-10-29:
It looks like Windows 8 doesn't recognize the serial port of the Model 100, so Chrysalis will be unable to communicate with it. I think it is possible to work around this by creating a `.INF` file to load the appropriate driver, but I'm not familiar with the details.

obra, 2022-10-31:
Realistically, given Windows 8 and 8.1's needs for specialized signed
drivers (not required by Windows 7 or Windows 10) even for serial devices
combined with the fact that we're 3 months from Microsoft's official end of
life for 8.1, I don't think we're ever going to be able to officially
support Chrysalis on Windows 8.

Best,
jesse

closed 2024-02-26 by obra