keyboardio/Chrysalis #1226: Keyboardio Model 100: Pause Does Not Pause Music

Zian, 2022-11-07:
**Describe the bug**
After I press the fun and enter keys on a stock Keyboardio Model 100, the music keeps playing.

**To Reproduce**
Steps to reproduce the behavior:
1. Buy a Keyboardio Model 100 using Kickstarter.
2. Plug the device in and update the firmware as instructed by Keyboardio.
3. Play music. For example, download and install the [K-lite Codec Full Pack](https://codecguide.com/download_k-lite_codec_pack_full.htm). Then, download [some music](https://www.youtube.com/watch?v=bL3vr9Dcmus) using [yt-dlp](https://github.com/yt-dlp/). Finally, launch MPC-HC and play the music.
4. Do something else like filling out the GitHub new issue form for a few seconds.
5. Press the fun key and hold it down while pressing the enter key.

**Expected behavior**
I expected the music to stop.

[chrysalis-debug.json.txt](https://github.com/keyboardio/Chrysalis/files/9947701/chrysalis-debug.json.txt)

**Desktop (please complete the following information):**
 - OS: Microsoft Windows Version 21H2 (OS Build 19044.2130)
 - Chrysalis Version: 0.12.0

**Additional context**
Current firmware version: 0.91.0+68

otstrel, 2023-01-22:
I just updated keyboard to latest firmware offered by Chrysalis checking "erase everything on my keyboard" box and now I'm having exactly the same problem - Play/Pause key no longer works.
It worked before firmware update.
Same key works fine as Enter so it is not a key issue.
Next/Prev Track keys also work fine. 
Mapping Play/Pause to any other key in any other layer doesn't work either.
I used trial version of [PassMark KeyboardTest](https://www.passmark.com/products/keytest/) to check what's going on (btw, I expected Chrysalis to have similar functionality for troubleshooting purposes) and pressing Play/Pause does not generate any key codes at all. 

[chrysalis-debug.json.txt](https://github.com/keyboardio/Chrysalis/files/10475111/chrysalis-debug.json.txt)

otstrel, 2023-01-26:
Strangely enough, reboot fixed my problem - Play/Pause works now and happily generates keycodes in KeyboardTest 🤷‍♂️

obra, 2024-11-11:
@Zian - is this still an issue? If so, please reopen the issue. I'm reliably able to pause and resume music.

closed 2024-11-11 by obra