keyboardio/Chrysalis #1230: Add programmer dvorak layour

alerque, 2022-11-10:
I've been user of the [programmer dvorak](https://www.kaufmann.no/roland/dvorak/) layout for over a decade. Given that using a layout like this means my keycaps never match my actual usage, but hey. That's why I bought Linear A keycaps for my two model 100s! They are fantastic.

That being said it's really confusing to use the keymap editor in Chrysalis to map keys to the actual layout I use. The DVORAK layout gets it about 70% of the way there but also makes the remaining 30% that much more confusing.

It would be lovely to see programmer dvorak added so that the keybord editor and layout cards could be accurate.

I might even be willing to chip in and contribute one, but am opening this issue first to track before even researching what that would take.

obra, 2022-11-10:
Indeed, the first step would be to build up the layout in Chrysalis and to export the layer json as an attachment on this issue. From there, @algernon can likely take it.

alerque, 2022-11-10:
I think this issue may be conflating two different things. There seem to be configured presets for programming a keyboard to *output* keycodes based on a given layout (without the host computer having to know about that layout) and a reference layout used in they keymap editor to show key labels and pick keys. I was refering to the later; @obra's comment appears to refer to the former. I base that on the former using JSON based layouts and being something you can build up in Chrysalis itself, the latter that I was talking about seems to be XML format.

It looks to me like starting with [static/cldr/keyboards/windows/en-t-k0-windows-dvorak.xml](https://github.com/keyboardio/Chrysalis/blob/master/static/cldr/keyboards/windows/en-t-k0-windows-dvorak.xml) and adapting to for the DVP adjustment.

obra, 2022-11-10:
Ah, indeed, if you're looking to make Chrysalis aware of a non-CLDR
host-side layout, that's going to be a relative lot of work. We don't yet
have any kind of provision for extending the CLDR layout database.

alerque, 2022-11-10:
Are the XML files auto generated from CLDR data?

obra, 2022-11-10:
They are