keyboardio/Chrysalis #1232: Setting colours on a layer has no effect

thommay, 2022-11-16:
**Describe the bug**
I've set colours on a number of keys on layer 2 to help me remember what they are. However the colours are not shown when I shift or lock to layer 2. 

**To Reproduce**
Steps to reproduce the behavior:
1. In layout editor, set colours on many keys on layer 2
3. save to keyboard
4. Shift to layer 2 
5. No colours are shown

**Expected behavior**
Keys with colours set on them show those colours

**Screenshots**
[image]

[chrysalis-debug.json.txt](https://github.com/keyboardio/Chrysalis/files/10021159/chrysalis-debug.json.txt)

**Desktop (please complete the following information):**
 - OS: macOS 12.4

 - Chrysalis Version: 0.12

**Additional context**
Add any other context about the problem here.

dj311, 2023-01-20:
I had this issue and have found a solution: your keyboard needs to be on "LED MODE" 14. You can to this by cycling through modes by pressing the "led" key. 

But I found this annoying/cumbersome since it's hard to keep track of where in the sequence your keyboard is. Instead you can set this mode to be your default at startup by going to "My Keyboard" under "Preferences"  in Chrysalis, then set "Default led mode" to 14. (I personally then unassigned the "led" key so I couldn't mess it up in the future).

I think it would be worth documenting this in the UI under the "Colors" part of the editor.

closed 2024-11-11 by obra

eirvandelden, 2025-01-02:
How do you setup the led mode for a Model01? It does not have an LED mode in preferences?